PhilipsHue/flutter_reactive_ble #582: Peripherals get stuck in state DeviceConnectionState.connection without triggering timeout

d-wolf, 2022-06-27:
**Describe the bug**

Currently we are developing an app where the user can swipe through his gallery of BLE devices (more or less like in an image gallery). When he stops swiping for a few hundred milliseconds we start to establish a connection with the selected device to enrich the view with device data.

While the connection is getting establish the user also can just swipe to the next device in the gallery, wich should cancel the ongoing connection (attempt) and establishes a new connection to the current selected device. In the end there should be just one single device in the connected state. All the others should be disconnected.

The problem starts when we are switching devices pretty fast and trying to connect another device before the previous selected has been connected/disconnected. For me it's seems that some devices get stuck in DeviceConnectionState.connecting without triggering a timeout.

**To Reproduce**
Steps to reproduce the behavior:
To reproduce the issue we have made a small sample application where the user can switch between 3 devices shown in a listview. The listview shows the last selected device (blue) and the current DeviceConnectionState of each device. When you hit different devices fast in a row, you can see that some connections will get stuck in DeviceConnectionState.connecting.

[image]

To get the sample app running change the mac addresses Strings of device1, device2 & device 3 to your own BLE devices around.

Code:

```Dart
import 'dart:async';

import 'package:flutter/material.dart';
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart';

void main() {
  runApp(const MyApp());
}

class MyApp extends StatelessWidget {
  const MyApp({Key? key}) : super(key: key);

  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      title: 'Flutter Demo',
      theme: ThemeData(
        primarySwatch: Colors.blue,
      ),
      home: const MyHomePage(),
    );
  }
}

class MyHomePage extends StatefulWidget {
  const MyHomePage({Key? key}) : super(key: key);

  @override
  State<MyHomePage> createState() => _MyHomePageState();
}

class _MyHomePageState extends State<MyHomePage> {
  final FlutterReactiveBle _ble = FlutterReactiveBle();

  static const Duration timeout = Duration(seconds: 5);

  static const String device1 = 'DD:3C:BD:BF:DA:FD';
  static const String device2 = 'F8:5E:C8:A9:71:D6';
  static const String device3 = 'C2:B9:28:FB:BA:5B';

  final List<String> _deviceIds = [
    device1,
    device2,
    device3,
  ];

  final Map<String, DeviceConnectionState> _connectionStateMap = {
    device1: DeviceConnectionState.disconnected,
    device2: DeviceConnectionState.disconnected,
    device3: DeviceConnectionState.disconnected,
  };

  final Map<String, StreamSubscription<ConnectionStateUpdate>?>
      _connectionSubscriptionMap = {
    device1: null,
    device2: null,
    device3: null,
  };

  int? _lastSelectedDeviceIndex;

  @override
  void initState() {
    super.initState();

    _ble.connectedDeviceStream.listen((event) {
      debugPrint('$event');
      setState(() {
        _connectionStateMap[event.deviceId] = event.connectionState;
      });
    });
  }

  @override
  Widget build(BuildContext context) {
    return Scaffold(
        body: ListView.separated(
      itemCount: _deviceIds.length,
      itemBuilder: (BuildContext context, int index) {
        return ListTile(
          selected: _lastSelectedDeviceIndex == index,
          title: Text(_deviceIds[index]),
          trailing: Text(
            _connectionStateMap[_deviceIds[index]].toString().split('.')[1],
          ),
          onTap: () async {
            setState(() {
              _lastSelectedDeviceIndex = index;
            });

            await _onConnect(_deviceIds[index]);
          },
        );
      },
      separatorBuilder: (BuildContext context, int index) {
        return const Divider();
      },
    ));
  }

  Future<void> _onConnect(String id) async {
    for (var connection in _connectionSubscriptionMap.values) {
      await connection?.cancel();
    }

    _connectionSubscriptionMap[id] = _ble
        .connectToDevice(id: id, connectionTimeout: timeout)
        .listen((event) {},
            onError: (e) => debugPrint(e), onDone: () => debugPrint('onDone'));
  }
}

````

**Expected behavior**
We would expect that only the last selected device is finally in DeviceConnectionState.connected. All other device should be in DeviceConnectionState.disconnected. The timeout should get triggered when the connection gets stuck in DeviceConnectionState.connected.

**Smartphone / tablet**
 - Device: Nokia 7.1
 - OS: Android 10
 - Package version: 5.0.2

**Peripheral device**
- Vendor, model: CUSTOM/ Nordic NRF
- Does it run a custom firmware / software: yes

d-wolf, 2022-06-27:
Could be connected to [this issue](https://github.com/PhilipsHue/flutter_reactive_ble/issues/558).

d-wolf, 2022-06-30:
After some more Tests i found out how to easily reproduce the issue:
1. connectToDevice A
2. When device A is in state connecting, cancel the connection to device A and connectToDevice B
3. **Device A and B will now stuck in state connecting**

https://user-images.githubusercontent.com/3867384/176616574-c9d57acd-84d1-4e81-90ec-4587ff5b4f70.mp4

ryanheitner, 2024-05-01:
I am also seeing stuck in connecting on Android without a time out