PhilipsHue/flutter_reactive_ble #586: subscribeToCharacteristic listen does not work on iOS

MeteGUL, 2022-06-28:
**Describe the bug**
First I want to start by saying that the library works fine on android. In my iOS version, I connect the device to the iPad without any problems. My problem starts here. After connecting my device, I use a button to start listening. SubscribeToCharacteristic works on android, no response on iOS. It neither goes into listen, nor goes into onError, nor does it go into onDone. I have read the issues opened on this subject at least 3 times, and I could not find any solution. I am attaching my code below.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to device using `connectTo()`.
2. Try to start, but no response.

**Expected behavior**
To be able to subscribe to device. Access the data.

**Smartphone / tablet**
 - Device: iPad 8th generation 
 - OS: [e.g. iOS 14.0.1]

Code:

  Uuid _UART_UUID = Uuid.parse("xxxxxxxx-0000-8000-5000-xxxxxxxxxxxx");
  Uuid _UART_RX = Uuid.parse("xxxxxxxx-0000-8000-5000-xxxxxxxxxxxx");
  Uuid _UART_TX = Uuid.parse("xxxxxxxx-0000-8000-5000-xxxxxxxxxxxx");

QualifiedCharacteristic _txCharacteristic, _txCharacteristicBattery;
  Stream<List<int>> _receivedDataStream, _receivedDataStreamBattery;
  StreamSubscription<List<int>> streamSubscription, streamBatterySubscription;
  final flutterReactiveBle = FlutterReactiveBle();

 Future<void> listenBattery(String dvcid) async 
{
    _txCharacteristicBattery =  QualifiedCharacteristic(serviceId: _UART_UUID, characteristicId: _UART_TX, deviceId: dvcid);

    _receivedDataStreamBattery = flutterReactiveBle.subscribeToCharacteristic(_txCharacteristicBattery);

    print("BATTERY ERROR before :");

/// I CAN'T SEE OTHER PRINTS AFTER THAT.

    streamBatterySubscription = _receivedDataStreamBattery.listen((data) {

        print("BATTERY ERROR before2 : ");
........
      },onError:(e) {

        print("BATTERY ERROR : ${e}");
    },onDone:() {

      print("BATTERY ONDONE");
    },
    );
  }

FurkanTskn, 2022-06-28:
i have same isssue

MeteGUL, 2022-07-01:
I ran it in the sample application. I printed the uuid well to understand the difference. The problem was the Android/iOS uuid difference. Now I can run it. If you're getting the error I got, I suggest you run the sample application and examine the uuids.

re-mike, 2022-07-06:
I had the same issue, subscribeToCharacteristic worked on Android but not on iOS.
In my case it was Uuid related issue as well.

I am using Bluetooth 16bit Uuids for my characteristics, e.g. 2a5a
In my Android app I was parsing the full length version of the Bluetooth Uuid:
`Uuid.parse('00002a5a-0000-1000-8000-00805f9b34fb');`
This works fine on Android but not on iOS.
I used the full length version initially because the discoverServices function returns the full length on Android.

To make it work on iOS, I had to parse it as just the 16bit Uuid:
`Uuid.parse('2a5a')`
This appears to work on both platforms.
The discoverServices function returns the 16 bit version on iOS.

rajkumar82, 2022-07-16:
I get a disconnect as soon as I subscribe to characteristics.

ConnectionStateUpdate(deviceId: FBE1632C-DCD5-33B1-0518-6EF6ACD13551, connectionState: DeviceConnectionState.disconnected, failure: GenericFailure<ConnectionError>(code: ConnectionError.failedToConnect, message: "Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.}"))

I tried Uuid.parse('xxxx'), it works only in android, but not in IOS.

MeteGUL, 2022-07-18:
> 

Can you try Uuid.parse('632C') ?
I solved it this way. If it doesn't resolve, give it a try in the sample app. If it works correctly, you can find the relevant Uuid part and write print there, you can reach the correct Uuid.

rajkumar82, 2022-07-19:
@MeteGUL 
Please note that the guid shown above is that of the device id
deviceId: FBE1632C-DCD5-33B1-0518-6EF6ACD13551

Both the below lines

final Uuid _UART_UUID = Uuid.parse("0000a002-0000-1000-8000-00805f9b34fb");
final Uuid _UART_RX = Uuid.parse("0000c306-0000-1000-8000-00805f9b34fb");
final Uuid _UART_TX = Uuid.parse("0000c304-0000-1000-8000-00805f9b34fb");

as well as these

  final Uuid _UART_UUID = Uuid.parse("a002");
  final Uuid _UART_RX = Uuid.parse("c306");
  final Uuid _UART_TX = Uuid.parse("c304");

work fine in Android.

I tried both versions of guid in ios and it does not work. 

I had also included WidgetsFlutterBinding.ensureInitialized(); in main as suggested in an another issue.

Also I have made sure I try to subscribe only after ensuring status is BleStatus.ready.

Few people have suggested to turn off 2.4GHz wifi or switch to 5Ghz Wifi network as this causes interference with BLE.

None of the above solves the issue.

Please suggest if there anything else that I need to try.

ben91187, 2023-04-05:
iOS restricts direct access to certain characteristics

https://developer.apple.com/forums/thread/35100