PhilipsHue/flutter_reactive_ble #597: Device StreamSubscription does not disconnect even with the "deviceSubscription.cancel()" method.

yuriboeira11tx, 2022-08-04:
**Describe the bug**
When I enter the screen of my application, there is a function responsible for connecting to the device. It works perfectly. The problem is when you just enter this screen and exit, even using "dispose()" and inside it canceling any connection StreamSubscription, it still remains connected. Try connecting and while connecting, press to disconnect. It's not closing. This is very strange as it is a Stream method.

**Smartphone / tablet**
 - Device: Samsung M62
 - OS: Android 12
 - Package version: 5.0.2

**Peripheral device**
- Does it run a custom firmware: yes

FrancoisB-HEX, 2024-11-05:
I have the same issue on Samsung S22+ Android SDK 34 with package **5.0.3**:
I disconnect the device by calling `cancel()` on the connection `Stream<ConnectionStateUpdate>`. I even see a final `ConnectionStateUpdate(deviceId: XYZ, connectionState: DeviceConnectionState.disconnected, failure: null)`, but the device is no longer discovered on scanning for devices. If I disable and re-enable Bluetooth on the phone, or I restart the BLE peripheral, I once again discover it and I can connect to it.

FrancoisB-HEX, 2024-11-06:
> I have the same issue on Samsung S22+ Android SDK 34 with package **5.0.3**: I disconnect the device by calling `cancel()` on the connection `Stream<ConnectionStateUpdate>`. I even see a final [...]

I can confirm updating the the current [HEAD](https://github.com/PhilipsHue/flutter_reactive_ble/commit/a6dd92648c4887c63320122dcfcd5c9dad1da40b) (and possibly v5.3.1) solves this issue - the connection is properly closed when cancelling the subscription.