PhilipsHue/flutter_reactive_ble #598: Two streams subscription in real time to different SubscribeToCharacteristic.

dawiddszewczyk, 2022-08-09:
**Describe the bug**
I would like to subscribe two services like "battery service " and "UART service" in realtime together to listen battery status and also get live data from device. But already I getting single live data.

**To Reproduce**
  Future<void> _onDeviceCallbackOne() async {
    _monitorLiveDataSubscription = RetryStream(() => flutterReactiveBle
            .subscribeToCharacteristic(QualifiedCharacteristic(characteristic))
            .doOnListen(() {
         print('Subscribed to LiveData');
        }).applyTransformer(transformer)).listen((event) {});

   _monitorLifeBatterySubscription = RetryStream(() => flutterReactiveBle
            .subscribeToCharacteristic(QualifiedCharacteristic(characteristicBattery))
            .doOnListen(() {
         print('Subscribed to Battery Life');
        }).applyTransformer(transformerBattery)).listen((event) {});
  }

**Smartphone **
 - Device: IPhone SE
 - OS: iOS 15
 - Package version: 5.0.2

ElreboCM, 2022-09-25:
This seems to be a duplicate of Issue #526.

closed 2022-10-24 by Taym95

ElreboCM, 2022-10-24:
@Taym95: Can you reason why you closed this issue? In my test, this issue is not solved yet (see also issue #526).

Taym95, 2022-10-24:
duplicate issue