PhilipsHue/flutter_reactive_ble #600: Scanning does not work, fails on lack of location permission

creikey, 2022-08-26:
**Describe the bug**
E/flutter (30619): [ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Location Permission missing (code 3)")

**To Reproduce**
```dart
void _refreshDevices() async {
    bool goForIt=false;
    if (Platform.isAndroid) {
      PermissionStatus locationPermission = await Permission.location.request();
      PermissionStatus finePermission= await Permission.locationWhenInUse.request();
      // TODO user feedback on no location
      if (locationPermission == PermissionStatus.granted && finePermission == PermissionStatus.granted) {
        goForIt=true;
      }
    } else if (Platform.isIOS) {
      goForIt=true;
    }

    if(goForIt) {
      setState(() {
        _devices = [];
      });
      widget.flutterReactiveBle.scanForDevices(withServices: [], scanMode: ScanMode.lowLatency).listen((device) {
        setState(() {
          _devices.add(device.name);
        });
      });
    }
  }
```
This code fails with an exception of not having location permissions even though they're granted and accepted. 

**Expected behavior**
If location permissions are granted, it doesn't except on location permissions not being granted

- [ x] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: [e.g. iPhone X]
 - OS: [e.g. iOS 13, Android 10]
 - Package version: [e.g. 1.0.0]
Oneplus 8 most updated

**Peripheral device**
n/a

**Additional context**
n/a

creikey, 2022-08-26:
Related: #51

creikey, 2022-08-26:
I needed to do all of these permissions:
```dart
PermissionStatus locationPermission = await Permission.location.request();
PermissionStatus bleScan = await Permission.bluetoothScan.request();
PermissionStatus bleConnect = await Permission.bluetoothConnect.request();
```

Would be nicer if the error message was more descriptive? It made it seem like only location was not authorized

closed 2022-08-26 by creikey

jaison-p-joy, 2023-10-16:
getting same error