PhilipsHue/flutter_reactive_ble #601: Duplicate response from writeCharacterisiticWithResponse

ASHBAW, 2022-09-02:
I have an app with two screens. When entering each screen it will subscribe to a characteristic and exiting it will cancel the subscription.

When I connect, I start out on screen one and am able to writeCharacterisiticWithResponse() continuously. I then go to another screen and am able to writeCharacterisiticWithResponse() with no issue. When I exit a screen I am canceling in the dispose method as follows:

```
  @override
  void dispose() {
    await subscribeStreamDataReady?.cancel();
    super.dispose();
  } 
```

When going back to the screen one from screen two I am seeing a duplicate response:

GOOD CASE
```
I/flutter (13873): Subscribing to: 93c7a770-ddff-11e4-b9d6-1681e6b88ec1
I/flutter (13873): BLE Write : [39, 1, 1, 0, 0, 11, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] to 737bea54-d3c5-11e4-b9d6-1681e6b88ec1
I/flutter (13873): BLE Response : [39, 0, 1, 1, 0, 0, 11, 49, 46, 49, 46, 54, 0, 0, 0, 0, 0, 0]
I/flutter (13873): BLE Write : [44, 1, 1, 3, 0, 8, 7, 230, 17, 21, 14, 2, 8, 0] to 737bea54-d3c5-11e4-b9d6-1681e6b88ec1
I/flutter (13873): BLE Response : [44, 0, 1, 1, 3, 0, 8, 7, 230, 17, 21, 14, 2, 8, 0]
I/flutter (13873): BLE Write : [32, 2, 1, 67, 73, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] to 737bea54-d3c5-11e4-b9d6-1681e6b88ec1
I/flutter (13873): BLE Response : [32, 0, 2, 1, 67, 73, 12, 102, 111, 111, 98, 97, 114, 0, 0, 0, 0, 0, 0]
I/flutter (13873): BLE Write : [32, 2, 2, 67, 73, 8, 0, 0, 0, 0, 0, 0, 0, 0] to 737bea54-d3c5-11e4-b9d6-1681e6b88ec1
I/flutter (13873): BLE Response : [32, 0, 2, 2, 67, 73, 8, 0, 0, 0, 0, 0, 0, 0, 0]
```
BAD CASE (SEE <== DUPLICATE BELOW)
```
I/flutter (13873): Subscribing to: 93c7a770-ddff-11e4-b9d6-1681e6b88ec1
I/flutter (13873): BLE Write : [39, 1, 1, 0, 0, 11, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] to 737bea54-d3c5-11e4-b9d6-1681e6b88ec1
I/flutter (13873): BLE Response : [39, 0, 1, 1, 0, 0, 11, 49, 46, 49, 46, 54, 0, 0, 0, 0, 0, 0]
I/flutter (13873): BLE Write : [44, 1, 1, 3, 0, 8, 7, 230, 5, 22, 14, 2, 8, 0] to 737bea54-d3c5-11e4-b9d6-1681e6b88ec1
I/flutter (13873): BLE Response : [39, 0, 1, 1, 0, 0, 11, 49, 46, 49, 46, 54, 0, 0, 0, 0, 0, 0] <== DUPLICATE
I/flutter (13873): BLE Write : [32, 2, 1, 67, 73, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] to 737bea54-d3c5-11e4-b9d6-1681e6b88ec1
I/flutter (13873): BLE Response : [32, 0, 2, 1, 67, 73, 12, 102, 111, 111, 98, 97, 114, 0, 0, 0, 0, 0, 0]
I/flutter (13873): BLE Write : [32, 2, 2, 67, 73, 8, 0, 0, 0, 0, 0, 0, 0, 0] to 737bea54-d3c5-11e4-b9d6-1681e6b88ec1
I/flutter (13873): BLE Response : [32, 0, 2, 1, 67, 73, 12, 102, 111, 111, 98, 97, 114, 0, 0, 0, 0, 0, 0]
I/flutter (13873): BLE Write : [32, 2, 1, 86, 73, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] to 737bea54-d3c5-11e4-b9d6-1681e6b88ec1
```
Any thoughts on why this is happening? If I disconnect from the device and reconnect it goes to screen one, starts subscription, writeCharacterisiticWithResponse() continuously with no issue every time. The issue comes from going to another screen and coming back to screen one.

yuriboeira11tx, 2022-09-05:
I noticed that the library has a problem with unsubscribing. Both for disconnecting from the device and for notifications. On iOS this works reasonably as it throws an exception. I've been trying to find an answer to this problem for a long time.

ASHBAW, 2022-09-06:
Thanks for the information.