PhilipsHue/flutter_reactive_ble #604: iOS: Unable to connect to Raspberry Pi with BlueZ 5.55-2.1+rpt1

FcoFabra, 2022-09-08:
**Bug description**
We are developing a C application using Bluez_inc library.
The advertising includes flags, services UUIDs, and device name.
There is no problem to connect using Windows or Android, only on iOS.

Bluetoothctl shows the following output when trying to connect
`[NEW] Device 4C:12:37:84:9E:5E 4C-12-37-84-9E-5E`
`[CHG] Controller E4:5F:01:0D:B5:B1 SupportedInstances: 0x05 (5)`
`[CHG] Controller E4:5F:01:0D:B5:B1 ActiveInstances: 0x00 (0)`
`[CHG] Device 4C:12:37:84:9E:5E Connected: no`
`[CHG] Controller E4:5F:01:0D:B5:B1 SupportedInstances: 0x04 (4)`
`[CHG] Controller E4:5F:01:0D:B5:B1 ActiveInstances: 0x01 (1)`
`[DEL] Device 4C:12:37:84:9E:5E 4C-12-37-84-9E-5E`

**To Reproduce**
We use `connectToDevice()` to connect to a single device, but the connection process fails:

**Expected behavior**
The iPhone should connect to the Raspberry Pi and read some data using the published services and characteristics.

- [ ] I Tried doing the same with Bluetooth LE Explorer (Windows), BLE Scanner (Android), and our own Android application and we are able to connect to the Raspberry Pi without any issue.

Thank you for your help.

Taym95, 2022-11-07:
Can you share logs from the app ? and code snippet please? so I can help !

FcoFabra, 2022-11-08:
Thank you for your feedback.
I forgot to close this issue. The problem was not related to Flutter Reactive BLE.
We defined services and characteristics in a non-standard way. Android and Windows don't care about that approach, but iOS does.
We used a 16 bits standard UUID for the service, but 128 bits custom UUIDs for the characteristics of the same service. This custom setup seems to be incompatible with iOS. We developed a workaround specific for our needs.

Now I find that the device disconnects after some time, but I could not be related again to Flutter Reactive BLE.

Thanks again.

closed 2022-11-08 by FcoFabra