PhilipsHue/flutter_reactive_ble #612: Fail to receive the TimeoutException after the given timeout duration

tc14077s, 2022-09-26:
It seems the app cannot receive the `TimeoutException` while the [connectionTimeout] parameter is already supplied to `connectToDevice`.  And, only the `onDone` handler can triggered.

**what I do** 
1. scan and get the ble device id
2. turn off the ble device(making it not connectable)
3. call connectToDevice with connectionTimeout value.

the example code:
```
_flutterReactiveBle.connectToDevice(
      id: bleDevice.id,
      connectionTimeout: Duration(seconds: 2),
    ).listen(
      (event) async {
        // Things to do after connect
      },
      onError: (Object error) {
       // expect to catch TimeoutException here, but the app never fall into this handler.
      },
      onDone: () {
        // this handler is triggered after the connection is close, also applies to the connection failure case
      },
    )
```

**Expected behavior**
- expect to receive `TimeoutException` from the Stream

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

I have do the test on both android (12) and iOS(14.5.1)  platform, both platforms are having same issue like what I explained above. For the android platform, there is an event (`DeviceConnectionState`) from Stream with a `failure` `ConnectionError.failedToConnect` status. Sadly, we don't have this failure message in iOS platform when doing this test case. Are there any ways to know if the connection cannot be established before `connectionTimeout` expires?

Pierre-Monier, 2023-03-28:
Same here! I would love to see this fixed :)

Brinfotech1407, 2023-06-26:
Same here! any working way you found @tc14077s

sdgroot, 2023-08-31:
Same here, the `onError` function is not called when a connection timeout occurs. Hope that this will get fixed soon!

tc14077s, 2023-09-15:
@Brinfotech1407 I implemented a timeout mechanism instead of relying on the package interface.

ramupal2603, 2023-09-15:
> @Brinfotech1407 I implemented a timeout mechanism instead of relying on the package interface.

Great can you explain a bit or some gist will be helpful.

Brinfotech1407, 2023-09-15:
> @Brinfotech1407 I implemented a timeout mechanism instead of relying on the package interface.

Great can you explain a bit or some gist will be helpful.