PhilipsHue/flutter_reactive_ble #614: Writing to multiple devices

santa010, 2022-10-06:
I am trying to send messages to multiple connected devices using `writeCharacteristicWithoutResponse` and I would expect a somewhat simultaneous change on the devices, but there seems to be a visible delay between execution on each device (enough that you can see a chain reaction from device 1 to device N). Does anyone know, if this is a limit of the library itself (i.e. the way it's designed or something similar)?

paintingStyle, 2022-10-17:
> I am trying to send messages to multiple connected devices using `writeCharacteristicWithoutResponse` and I would expect a somewhat simultaneous change on the devices, but there seems to be a [...]

I also have this problem，did you solve it

I want to ask a question. Due to the company's business, there are 5 bluetooth devices. I need to connect 5 devices at the same time, and get the data and update data of the device at the same time. I checked the information on the Internet and found that Android needs to send data serially.

https://stackoverflow.com/questions/28016571/robustly-communicating-with-multiple-ble-devices-simultaneously-on-android

Is there a way to write and update the characteristics of multiple devices at the same time?

Taym95, 2022-10-24:
I dont think Ble works like this, from doc `BluetoothGatt object you should only execute one operation at a time`.

closed 2022-10-24 by Taym95