PhilipsHue/flutter_reactive_ble #620: [Question]: Can the mobile terminal connect and process data from multiple Bluetooth devices?

paintingStyle, 2022-10-17:
I want to ask a question. Due to the company's business, there are 5 bluetooth devices. I need to connect 5 devices at the same time, and get the data and update data of the device at the same time. I checked the information on the Internet and found that Android needs to send data serially. 

Is there a way to write and update the characteristics of multiple devices at the same time?

Taym95, 2022-10-26:
Hi, this library doesn't support write and update the characteristics of multiple devices

closed 2022-10-26 by Taym95

paintingStyle, 2022-11-14:
> 嗨，这个库不支持写入和更新多台设备的特性

Thanks for your reply, is there any way I can improve the speed of getting device data,
The requirement is: get the measurement values of 4 kinds of data on one page.
My current thinking is:
Manual coding implementation, linking four devices ABCD at the same time, executing commands in sequence, and getting the measured values of the devices, I don't know if this is technically feasible? Hope to get your answer

paintingStyle, 2022-11-14:
> > 嗨，这个库不支持写入和更新多台设备的特性
> 
> 谢谢你的回复，我有什么办法可以提高获取设备数据的速度吗？要求是：在一页上获取4种数据的测量值。我目前的想法是：手动编码实现，同时链接四个设备ABCD，按顺序执行命令，并获得设备的测量值，我不知道这在技术上是否可行？希望得到你的答案

> > 嗨，这个库不支持写入和更新多台设备的特性

Thank you for your reply, is there any way to improve the speed of getting device data?
————————————
The requirement is: one page can get the measurement values of 4 kinds of devices.
—————————————
My current idea 1 is: Manually code the implementation, link four devices ABCD at the same time, execute commands in sequence, and get the measured values of the devices, I don't know if this is technically feasible? Hope to get your answer
————————————
My current idea 2 is: link device A first. Close the connection after getting the data, and then get the value of the BCD device in turn