PhilipsHue/flutter_reactive_ble #621: Remove device when powered off

pmagnuson, 2022-10-20:
**Is your feature request related to a problem? (please describe)**
The discovered device list is not updated when a device is turned off.   Thus the user is presented with devices that are not really available.  

**Describe the solution you'd like**
I would like devices to be removed from the discovered devices when they are no longer advertising.

**Describe alternatives you've considered**
Considering adding a heuristic to the ble_scanner.dart file from example.  Seems necessary to have a periodic timer that scans the _devices list to look for devices with an old "last seen" time.  

**Additional context**
Add any other context or screenshots about the feature request here.

chandlery54, 2024-07-23:
Any example of how to do this? Because the example app does not do this, and the only way I can think of it is like mentioned, scanning again, but I think stopping and starting the scan over and over seems a little bit ridiculous. There has got to be a better way.

GoodOldBatman, 2024-07-24:
It probably makes sense to just keep scanning when the user actually wants to see the available devices. If you want this throughout the whole runtime of the appp you actually can scan indefinitely while the app is running (on most devices) if you add short stops every ~30 s. But as you said this would be a bit of an overkill just to keep a list up to date that the user looks at just for 15 seconds and then continues with something else.

So when the user actually looks at the list of devices, you scan until he makes a decision. You will get repeated results of the DiscoveredDevice with the same identifier (as long as it's advertising). Remove it after e.g. 5 seconds of no result for example with a ``Timer`` for each identified device

    Map<String, DiscoveredDevice> devices ={
      // 'XX:XX:XX:XX:XX:XX': discoveredDeviceWithIdentifierA,
      // 'YY:YY:YY:YY:YY:YY': discoveredDeviceWithIdentifierB,
    };
    Map<String, Timer> timers = {
      // 'XX:XX:XX:XX:XX:XX': timerDeviceA,
      // 'YY:YY:YY:YY:YY:YY': timerDeviceB,
    };
  
    void onScanResult(DiscoveredDevice device){
      timers[device.id]?.cancel();
      timers[device.id] = Timer(const Duration(seconds: 5), () {
        devices.remove(device.id);
        timers.remove(device.id);
        // update ui
        notifyListeners(); // e.g. with Changenotifier
      });
      devices[device.id] = device;
      // update ui
      notifyListeners(); // e.g. with Changenotifier
    }

chandlery54, 2024-07-24:
Awesome. Thank you @GoodOldBatman for the example. That should definitely help. I just need to show a small list of BLE devices that are available to connect to for a few seconds until the user selects a choice, but if another user selects it first then it should disappear from other lists that may be scanning nearby at the same time. So this timer idea should help clear out devices that are not available anymore. I appreciate the help!