PhilipsHue/flutter_reactive_ble #624: Random disconnection with iOS device

mad-mikey, 2022-10-27:
**Describe the bug**
Bluetooth disconnection from iOS device and esp32. I would like to run an OTA updating using BLE. The first connection works, I send with writeWithoutResponse the packet size and it works, but when I send a command to start OTA after some time I have the disconnection

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device using `connectTo()`
2. Subscribe to a non-protected characteristic in listen mode
3. Write to a non-protected characteristic the packet size
4. Write to a non-protected characteristic the command to start the OTA
5. Observe a disconnection
6. Receive the notification ('ACK') in the listen (point 2)
7. Observe a failure with exception caused by the receiving of the notification ('ACK') in the listen (point 2) obviously because there's no more connection (Error unsubscribing from notifications: PlatformException(reactive_ble_mobile.Central.(unknown context at $10379fa50).Failure:1, The operation couldn’t be completed. (reactive_ble_mobile.Central.(unknown context at $10379fa50).Failure error 1.), {}, null))

**Expected behavior**
I expect no disconnection, in fact sometimes it works properly (only a very small number of times)
NOTE: NO PROBLEMS with Android

- I tried doing the almost the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above. Almost because in my application I tried both in a programmatically way and with buttons like in nRF Connect

**Smartphone / tablet**
 - Device: iPad (5th generation)
 - OS: iOS 15.5

**Peripheral device**
- Esp32
- Does it run a custom firmware: yes

**Additional context**
I have no problems with Android (same firmware, same app)

mad-mikey, 2023-03-31:
Nope, I can't find anything

Berkant35, 2023-05-05:
We first figure the set delay for each sent byte of data. This action helps us but not completely figure it out because this delay work only for iPhone 8 real device. After then we think to create a response message by esp32 and this response message each sends 509-byte data and then gets a response in the app. When the app sees this response message after sending a new 509 byte so this mean you can send new 509 byte data. This solution is very slow but we have successful results on many phones.

mad-mikey, 2023-05-18:
I understand, thank you.
My problem is that I can't wait so long, as a matter of fact I also tried (before post the first thread) to use the writeWithResponse but as you can see it takes too time.
Thank you again