PhilipsHue/flutter_reactive_ble #629: The device never finishes connecting and sometimes disconnects.

ChampiMagic, 2022-11-11:
**Describe the bug**
When I scan for my device it is easily found.

[image]

then I stop the scan and try to connect to the device with "connectToAdvertisingDevice" and the device stays in the "connecting" state and then disconnects (other times it gets stuck in the "connecting" state and does not advance).

[image]

My two functions:

[image]

[image]

Does anyone have any idea what could be going on?

SignSpice, 2022-11-17:
Have the same problem with the default example app.  No idea what is causing it, however.

ChampiMagic, 2022-11-18:
> Have the same problem with the default example app. No idea what is causing it, however.

I think I found the bug. The problem is that the device never finishes disconnecting properly so when it tries to connect back it receives the "action" to disconnect that it "never" received(or read) so it disconnects. The way to solve it is to restart the app (close and open it in production) or restart the BLE device.

ILittleNiZi, 2022-11-29:
I also encountered the same problem, and found that the example application also has the same problem

cyborg99, 2022-12-05:
I faced the same problem.
After disconnecting, when I try to connect again, the status remains "connecting".
However, adding the following to the esp32 source code solved the problem.

```c++
BLEServer *pServer = NULL;
bool deviceConnected = false;
bool oldDeviceConnected = false;

void setup()
{
  pServer->setCallbacks(new MyServerCallbacks());
  BLEDevice::startAdvertising();
  BLEDevice::startAdvertising();
}

class MyServerCallbacks : public BLEServerCallbacks
{
  void onConnect(BLEServer *pServer)
  {
    deviceConnected = true;
    Serial.println("onConnect");
  };
  void onDisconnect(BLEServer *pServer)
  {
    deviceConnected = false;
    Serial.println("onDisconnect");
  }
};

void loop()
{
  if (!deviceConnected && oldDeviceConnected)
  {
    delay(500);
    Serial.println("restartAdvertising");
    pServer->startAdvertising();
    oldDeviceConnected = deviceConnected;
  }
  if (deviceConnected && !oldDeviceConnected)
  {
    oldDeviceConnected = deviceConnected;
  }
}
```

farr64, 2022-12-05:
@cyborg99 Where, exactly, is the `esp32` source code?

Thanks

farr64, 2022-12-05:
Thanks, @cyborg99 .  I can now see:

ESP32でBLEデバイスを作る

(Make a BLE device with ESP32).

Are you by any chance working with a La Marzocco espresso machine? 

Anyhow, thanks for the reference.

real-geeno, 2023-01-25:
Has anyone found a workaround while waiting for the development team to fix the problem?

seanhamstra, 2023-04-12:
We've encountered the same on rare occasion. We'll receive the "connecting" event in the stream but never receive "connected" or any other state. Any updates here?