PhilipsHue/flutter_reactive_ble #631: dicoverServices on Android does not return dynamically added services

BikeBr0, 2022-11-17:
**Describe the bug**
I have a peripheral that adds an additional service after my flutter app connects and writes a value to a characteristic. Calling discoverServices on Android does not return the additional service but this works as expected on iOS.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one peripheral using connectToAdvertisingDevice()`
2. Call discoverServices and receive 3 Service Ids
3. Write to a protected characteristic
4. Call discoverServices again and receive the same 3 Service Ids
5. When trying to use the expected service, an exception is thrown: 

`_Exception (Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "Characteristic not found with UUID ________"))`

**Expected behavior**
The last call to discoverServices should return 4 services.

- [ X] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Google Pixel 6
 - OS: Android 14
 - Package version: 5.0.2

**Peripheral device**
- Vendor, model: CUSTOM
- Does it run a custom firmware / software: yes

lucasrutz, 2023-08-01:
same error

jml90, 2024-09-28:
Same error