PhilipsHue/flutter_reactive_ble #632: writeCharacteristicWithoutResponse fails on iOS

FXschwartz, 2022-11-22:
**Describe the bug**
After pairing to device, and discovering services. When I attempt to send a command to the sensor the writeCharacteristicWithoutResponse fails with the following error:
```
[VERBOSE-2:dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(reactive_ble_mobile.Central.(unknown context at $1061bb1a0).Failure:3, The operation couldn’t be completed. (reactive_ble_mobile.Central.(unknown context at $1061bb1a0).Failure error 3.), {}, null)
#0      StandardMethodCodec.decodeEnvelope
package:flutter/…/services/message_codecs.dart:653
#1      MethodChannel._invokeMethod
package:flutter/…/services/platform_channel.dart:296
<asynchronous suspension>
#2      new Stream.fromFuture.<anonymous closure> (dart:async/stream.dart:247:17)
<asynchronous suspension>
flutter: \^[[38;5;196m┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────<…>
flutter: \^[[38;5;196m│ \^[[0m\^[[39m\^[[48;5;196mException: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "A service 6E400002-B5A3-F393-E0A9-E50E24DCCA9E is not found in the peripheral 490582D6-5B9D-FFD3-CE73-DD653C8249BA (make sure it has been discovered)")<…>
```
I'm sure this is something I'm doing. With some context I am attempting to migrate our application from using flutter_blue_plus to this library.

I've tried adding a delay before subscribing to the characteristics as well tried writeCharacteristicWithResponse. It all works perfectly fine on Android just not iOS.

Any help would be greatly appreciated!

**Smartphone / tablet**
 - Device: iPhone 14 pro max
 - OS: iOS 16
 - Package version: v5.0.2

**Peripheral device**
- Vendor, model: CUSTOM
- Does it run a custom firmware / software: yes

**Additional context**
Add any other context about the problem here.

FXschwartz, 2022-12-13:
@remonh87 Hey Remon, I'm sure you are super busy, especially with end-of-the-year deadlines, but when you get a moment, can you take a look at this issue? Currently, it's the only thing keeping one of my client's projects from being migrated to use this package.

mashedbymachines, 2023-01-03:
I also have this problem with a product where we want to send a firmware file over BLE. Using "with response" as a workaround at the moment but it severly decreases throughput. Would be grateful if this was looked into

FXschwartz, 2023-01-03:
> I also have this problem with a product where we want to send a firmware file over BLE. Using "with response" as a workaround at the moment but it severly decreases throughput. Would be grateful if this was looked into

So using the method `writeCharacteristicWithResponse` works for you? Because neither works for me.

mashedbymachines, 2023-01-03:
> So using the method `writeCharacteristicWithResponse` works for you? Because neither works for me.

Yes on iOS the writeCharacteristicWithResponse method works bit not the one without response. On Android both methods work fine. 

However I now see that our error message is different, it is  just giving a write permission error when writing without response for us.

I have this problem:
https://github.com/PhilipsHue/flutter_reactive_ble/issues/360

ILittleNiZi, 2023-02-08:
have you solve this problem? i also have this problem

FXschwartz, 2023-02-08:
> have you solve this problem? i also have this problem

No, I am still blocked from migrating to this package because of this issue.

aabdulat1998, 2023-02-14:
I know this isnt what youre asking but maybe try the following. I set up a subscription stream to the tx characteristic first and then I writeCharacteristicWithoutResponse to the rx characteristic.

nikolayDemirev, 2023-03-08:
Declaring the characteristic on the BLE server with a write-without-response flag resolved the issue for me.