PhilipsHue/flutter_reactive_ble #635: How the write/read really working with the bytes?

MayaBloodsAI, 2022-11-27:
**Describe the bug**
It's not really a bug, maybe just something needed to be added to the document..
I need to write a stream of bytes to device so I was expected to send a parameter of **Uint8List**  but as I can see the **writeCharacteristicWithResponse** function get **List<int>** , does it mean that each item on the list is 4 bytes of data? I'm really confusing :) How should I send an stream of bytes?

MayaBloodsAI, 2022-11-28:
After play a little bit with the commends I understood.. Each "byte" represent by an item in the list of int - I guess my background in c just confused me :)

closed 2022-11-28 by MayaBloodsAI

TheauLep, 2023-02-07:
Hi, could you please help me ? I am facing the same issue, I don't really understand how to write the bytes I want. 

Thanks in advance

MayaBloodsAI, 2023-02-07:
Byte is between 0-255, just make sure the int in the List<int> is in this range

pvaibhav, 2023-02-20:
I really suggest describing this more in the documentation. It's not clear that the API expects `List<int>` to be a list of bytes (8-bit unsigned integers) even though `int` is a 64-bit signed integer. If your characteristic accepts signed integers, it is an easy mistake to make to just pass that signed integer as a single element inside the list, instead of properly building the byte buffer.