PhilipsHue/flutter_reactive_ble #649: Can't build with 5.0.3: The getter 'Uint8List' isn't defined for the class 'ProtobufConverterImpl'.

wabash9000, 2022-12-08:
I had a build working with 5.0.2 and then had to modify my pubspec.yaml to add an image asset to the build and it upgraded me to 5.0.3 and now I can't get my project to build. I tried removing 5.0.3 and setting pubspec to not upgrade from 5.0.2 and now there are errors with 5.0.2 that didn't happen before.

5.0.3 errors:
/C:/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.3/lib/src/converter/protobuf_converter.dart:49:42: Error: The getter 'Uint8List' isn't defined for the class 'ProtobufConverterImpl'.
 - 'ProtobufConverterImpl' is from 'package:reactive_ble_mobile/src/converter/protobuf_converter.dart' ('/C:/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.3/lib/src/converter/protobuf_converter.dart').
Try correcting the name to the name of an existing getter, or defining a getter or field named 'Uint8List'.
      message.serviceData.map((entry) => Uint8List.fromList(entry.data)),
                                         ^^^^^^^^^
/C:/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.3/lib/src/converter/protobuf_converter.dart:63:29: Error: The getter 'Uint8List' isn't defined for the class 'ProtobufConverterImpl'.
 - 'ProtobufConverterImpl' is from 'package:reactive_ble_mobile/src/converter/protobuf_converter.dart' ('/C:/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.3/lib/src/converter/protobuf_converter.dart').
Try correcting the name to the name of an existing getter, or defining a getter or field named 'Uint8List'.
          manufacturerData: Uint8List.fromList(message.manufacturerData),
                            ^^^^^^^^^
/C:/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.3/lib/src/converter/protobuf_converter.dart:61:24: Error: The argument type 'Map<Uuid, dynamic>' can't be assigned to the parameter type 'Map<Uuid, Uint8List>'.
 - 'Map' is from 'dart:core'.
 - 'Uuid' is from 'package:reactive_ble_platform_interface/src/model/uuid.dart' ('/C:/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_platform_interface-5.0.3/lib/src/model/uuid.dart').
 - 'Uint8List' is from 'dart:typed_data'.
          serviceData: serviceData,
                       ^

I get a similar Uint8List error in 5.0.2 now.

wabash9000, 2022-12-08:
I guess I should add context, trying to build for Android from Android Studio, but I also build for iOS, but haven't tried since this issue popped up.

wsmzdyss, 2022-12-12:
same problem，my solution is to lock 5.0.2 and clear cache.

wsmzdyss, 2022-12-13:
@SungchangChoi you need clear cache, I remove project from Android Studio and reopen it.

wabash9000, 2022-12-13:
flutter_reactive_ble was still throwing errors after I locked it to 5.0.2 in my pubspec.yaml and ran 'Invalidate Cache / Restart...' in Android Studio. I had to open the flutter_reactive_ble files and modify the pubspec.yaml in the plugin because it had references to reactive_ble_mobile and reactive_ble_platform_interface both of which had the ^ on them so it was updating them and throwing errors as well.

wsmzdyss, 2022-12-14:
You also need to enter the /flutter/.pub-cache/hosted/xxxx/ folder and delete the cache of version 5.0.3

rfuerst87, 2022-12-14:
If anyone struggles with the same issue, you need to upgrade flutter to version `3.3.0` or higher.

Liberations, 2022-12-28:
> rsion `3.3.0`

pubspec.lock modify version to 5.0.2

NavneetTechlene, 2022-12-28:
I have upgraded to flutter 3.3.3  now i am not having thiss issue

vontdeux, 2023-04-01:
i am on 3.3.9 and im having this issue.

mateuszwojtczak, 2023-04-18:
@vontdeux It's because of the way pubspec resolves transitive dependencies. I had the same issue and had to add dependency_override for `reactive_ble_platform_interface: 5.0.2`. Since the main package can still use caret syntax, even if you constrain `flutter_reactive_ble` version, it still might resolve its dependency to a newer version that's not compatible.

closed 2024-05-22 by Taym95