PhilipsHue/flutter_reactive_ble #650: Dematerialized: Scanning parameters are not set

AndreaBalde, 2022-12-12:
**Describe the bug**
This issue happens after calling disconnect function.
The following exception is thrown:
`
GenericFailure (GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Scanning parameters are not set"))
`

The exception is thrown by result.dart
[image]

inside the dematerialize function:
[image]

After the exception is thrown the ble scanner got freezed and you cannot connect again.

Note: We encounter this issue after implementing firebase messaging, using its background top-level function.
If we comment 
`FirebaseMessaging.onBackgroundMessage(callback)`
this exception is not throw.

**To Reproduce**
Steps to reproduce the behavior:
1. Implement FirebaseMessaging background event
2. Connect to device
3. Send BLE Message
4. Disconnect from the device

**Expected behavior**
Success disconnection and ready to connect again

**Smartphone / tablet**
 - Device: Xiaomi mi9, pixel 6
 - OS: Android 11, Android 13
 - Package version: flutter_reactive_ble 5.0.2

**Peripheral device**
- Vendor, model: nrf52832
- Does it run a custom firmware / software: yes

AndreaBalde, 2024-03-08:
Any update?

I still have the same problem and I've not found anything to fix it. 
The problem seems to happen when I run the scan, but it randomly throws an exception, and sometimes work successfully

riccardocescon, 2024-04-04:
@Taym95 @remonh87 is there any update on this issue?
It is still happening to me too.
```
I/flutter (11188): Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Scanning parameters are not set")
I/flutter (11188): #0      Result.dematerialize.<anonymous closure> (package:reactive_ble_platform_interface/src/model/result.dart:22:13)
I/flutter (11188): #1      Result.iif (package:reactive_ble_platform_interface/src/model/result.dart:34:21)
I/flutter (11188): #2      Result.dematerialize (package:reactive_ble_platform_interface/src/model/result.dart:15:28)
I/flutter (11188): #3      DeviceScannerImpl.scanForDevices.<anonymous closure>.<anonymous closure> (package:flutter_reactive_ble/src/device_scanner.dart:55:49)
I/flutter (11188): #4      _MapStream._handleData (dart:async/stream_pipe.dart:213:31)
I/flutter (11188): #5      _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
I/flutter (11188): #6      _rootRunUnary (dart:async/zone.dart:1407:47)
I/flutter (11188): #7      _CustomZone.runUnary (dart:async/zone.dart:1308:19)
I/flutter (11188): #8      _CustomZone.runUnaryGuarded (dart:async/zone.dart:1217:7)
I/flutter (11188): #9      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
I/flutter (11188): #10     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
I/flutter (11188): #11     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:1
```
i am using flutter_reactive_ble: ^5.3.1
This issue sometimes appears when starting the first scan, but almost always is fired when starting a second scan (while the first one is still active)