PhilipsHue/flutter_reactive_ble #654: Optional setting of setLegacy() to prevent gaps in scan results from legacy advertisers on Android

anthonyhunter, 2022-12-14:
5.0.3 adds .setLegacy(false) to scan settings by default, while not a bug, this setting can cause long gaps in scan results from fast advertising legacy devices on Android and changes behaviour of our app which listens to adv. beacons.

In 5.0.2 without setLegacy() our app would receive 10-20 adv per second continuously without gaps, in 5.0.3 we see 10-20 adv per second for 7-10 seconds followed by 2-4 seconds of no results (repeated)

Consider providing a means to optionally set legacy adv. in scan setting.

danielstuart14, 2022-12-15:
Hello @anthonyhunter, which smartphone model(s) are you testing with? This seems to be a bug with the BLE stack for this particular model, as I haven't found this behaviour with any of the devices I tested (S22+, Pixel 5, Moto G8, Pocofone f1).

While I think the default behaviour on android should match iOS (return both legacy and extended advertisings), we could add an option to disable extended advertising on android. As setLegacy is called by the scanSettings builder, maybe we could add an optional parameter to the scanForDevices function (Example: onlyLegacy: false)

anthonyhunter, 2022-12-16:
Hi @danielstuart14, I'm seeing behaviour on an A52 (Android 13 latest fw), haven't tried any other models yet. It does occur with two different client devices, one a nRF52832 beacon (adv 160ms) and the other a nRF52840 padlock (adv 40ms). No issue with iOS so assume it must be something Samsung specific, have tried overriding power saving - no luck