PhilipsHue/flutter_reactive_ble #658: The app can't start the scanning process without location permissions.

fufylev, 2022-12-15:
**What's wrong:**

I set all setting for AndroidManifest in accordance with your example.
```
<uses-permission android:name="android.permission.INTERNET" />
    <!-- required for API 18 - 30 -->
    <uses-permission
            android:name="android.permission.BLUETOOTH"
            android:maxSdkVersion="30" />
    <uses-permission
            android:name="android.permission.BLUETOOTH_ADMIN"
            android:maxSdkVersion="30" />

    <!-- required for API 23 - 30 -->
    <uses-permission-sdk-23
            android:name="android.permission.ACCESS_COARSE_LOCATION"
            android:maxSdkVersion="30" />
    <uses-permission-sdk-23
            android:name="android.permission.ACCESS_FINE_LOCATION"
            android:maxSdkVersion="30" />

    <!-- API 31+ -->
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission
            android:name="android.permission.BLUETOOTH_SCAN"
            android:usesPermissionFlags="neverForLocation" />
```
However, every time I start an application I see a screen with the following:
"Authorise the FlutterReactiveBle example app to use Bluetooth and location", 
what means that I hit the condition of
```
case BleStatus.unauthorized:
        return "Authorize the FlutterReactiveBle example app to use Bluetooth and location"
 ```
in the file `ble_status_screen.dart`.

I also tried to set the parameter `requireLocationServicesEnabled` to `false` in `scanForDevices` method in file `ble_scanner.dart`. Nevertheless any attempts to reach the scan screen are not successful.
But, if I ask for a LOCATION permission and give an access, then the app and scan works normally as requested, all peripherals are on the list.

Is there any solution on how to bypass LOCATION PERMISSION to work with scan?

**To Reproduce**
Steps to reproduce the behavior:
1. Install 5.0.3 version 
2. Set all as in an example
3. Start an app

**Expected behavior**
App starts and scan works without ask for a LOCATION PERMISSION

**Smartphone**
 - Device: Samsung S9
 - OS: Android 10
 - Package version: 5.0.3

Jurgendelaat, 2022-12-15:
Have you enabled the permissions on your device itself? (On your phone: "Settings -> Apps -> your_app_name -> permissions")

Without a package like [permission_handler](https://pub.dev/packages/permission_handler), you probably have to enable this yourself.

fufylev, 2022-12-17:
What I meant is that I don't want to give an access to the location at all. My app doesn't need a location service, so I want to skip this permission and use the BLE scanner only.

Taym95, 2022-12-20:
You can not, ble scanning allows your app to get user location, only in android 12 > you can remove it!

closed 2022-12-20 by Taym95