PhilipsHue/flutter_reactive_ble #660: Is it possible to get all bytes from ScanRecord on Android?

GsMoMo, 2022-12-20:
**Is your feature request related to a problem? (please describe)**
I'm trying to use this library to collect the manufacturer data on iOS and Android. But I noticed the received manufacturer data from the same device is quite different on iOS and Android. On iOS, I'm able to get the complete raw data. So is there any way to get a byte array of ScanRecord.getBytes() on Android?

**Describe the solution you'd like**
We could have another property "rawData" in the DiscoveredDevice.

**Describe alternatives you've considered**
N/A

**Additional context**
Thank you for the time to improve this library and make it work smoothly on iOS and Android.

PranayCelllabs, 2022-12-22:
Hi,
I faced the same issue. My solution was to manually set MTU size to max (247).

When, trying to connect to device. inside the listener (when you get the state as connected) call requestMtu with mtu as 247.

Example code: 

flutterReactiveBle.connectToDevice(id: deviceId,)
    .listen((ConnectionStateUpdate connectionState) {
        if (connectionState.connectionState.name == 'connected') {
          flutterReactiveBle.requestMtu(deviceId: deviceId, mtu: 247);
          // your functionality.
        } 
    }
);

This should give you whole data on android. Hope this helps.

GsMoMo, 2022-12-23:
> Hi, I faced the same issue. My solution was to manually set MTU size to max (247).
>
> When, trying to connect to device. inside the listener (when you get the state as connected) call requestMtu with [...]

Hi,
Thank you for the information 😁. But I'm trying to get the complete information from an unconnectable BLE device (Beacon broadcasting). Therefore, setting MUT to a larger size is not available to me. 😅

ray007, 2024-02-15:
If not the whole `ScanRecord`, I'd be interested in the `TXPOWER` value from there.
Didn't find a way to currently get that...