PhilipsHue/flutter_reactive_ble #664: 5.0.3 version does not can't scaan

fufylev, 2022-12-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one / many devices using `connectTo()` / `connectToAdvertisingDevice()`
2. Read from / write to / subscribe to a protected / non-protected characteristic
3. Observe a failure with exception (including the part of the stack trace, belonging to this package) …

**Expected behavior**
A clear and concise description of what you expected to happen.

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: [e.g. iPhone X]
 - OS: [e.g. iOS 13, Android 10]
 - Package version: [e.g. 1.0.0]

**Peripheral device**
- Vendor, model: [e.g. Fitbit Versa, iPhone SE, or CUSTOM]
- Does it run a custom firmware / software: yes / no

**Additional context**
Add any other context about the problem here.

watsmyname, 2022-12-29:
It works in my Android 12, not using code from example folder though.

PlayCulture, 2023-01-02:
I have an Android 11 phone here that can scan for iBeacons and an Android 13 phone that cannot.

PlayCulture, 2023-01-02:
I have observed the following, running identical code:
- Android 11 phone can detect AltBeacons and iBeacons
- Android 13 phone cannot

I've also seen the same behaviour with other beacon-sensing packages for flutter. Is there a possibility this is a permission-realted issue?

PlayCulture, 2023-01-03:
Update:
- Android 13 phone can detect AltBeacons, but **not iBeacons**

JulianAtTroido, 2023-01-05:
Based on this [Stackoverflow](https://stackoverflow.com/a/62473780) it seems that the underlaying Android library [RxAndroidBle](https://github.com/dariuszseweryn/RxAndroidBle) which flutter_reactive_ble uses defines the scanForDevices without on empty ScanFilter: see [here](https://github.com/PhilipsHue/flutter_reactive_ble/blob/4b5ed35fef136569c1efa56aa096919c839b9101/packages/reactive_ble_mobile/android/src/main/kotlin/com/signify/hue/flutterreactiveble/ble/ReactiveBleClient.kt) line 70.

PlayCulture, 2023-01-06:
Thanks again Julian - I'm looking forward to testing. It sounds like "flutter_reactive_ble breaking iBeacons for everything" might deserve its own issue thread? I'm curious what could have changed from Android 11 to cause an issue like this.

JulianAtTroido, 2023-01-06:
We just fixed our "problem" by using the solution from this  https://github.com/PhilipsHue/flutter_reactive_ble/issues/566#issuecomment-1181665284. 🎉
It is also outlined in the readme of this library.
Keep in mind, that you might need to remove `tools:targetApi="s"` as compared to the readme example.

Please tell me if that also resolved your problem.

PlayCulture, 2023-01-25:
Thanks again for the advice, much appreciated. Unfortunately the only thing that has worked for me is to completely remove reactive_ble. Now other ble packages can sense iBeacons again, so i'm up and running. Damn!

fufylev, 2023-01-25:
> We just fixed our "problem" by using the solution from this [#566 (comment)](https://github.com/PhilipsHue/flutter_reactive_ble/issues/566#issuecomment-1181665284). 🎉 It is also outlined in the readme of this library. Keep in mind, that you might need to remove `tools:targetApi="s"` as compared to the readme example.
> 
> Please tell me if that also resolved your problem.

It did not help me. I still can't detect iBeacons on Android 12

fufylev, 2023-01-25:
> Thanks again for the advice, much appreciated. Unfortunately the only thing that has worked for me is to completely remove reactive_ble. Now other ble packages can sense iBeacons again, so i'm up and running. Damn!

Which package you've chosen?

PlayCulture, 2023-01-25:
I'm using flutter_beacon - As of today we have broadcasting and sensing working on Android 11 and 13, as well as iOS. Hope that helps!

fufylev, 2023-01-25:
> I'm using flutter_beacon - As of today we have broadcasting and sensing working on Android 11 and 13, as well as iOS. Hope that helps!
> 
> Top tip that might help you: iOS handles bluetooth permission weirdly, so you don't want to explicitly request it, just go ahead and try to use bluetooth and it will automatically request permission. For Android you should request first, then use

Thanks for reply.
I've spent several hours on this flutter_beacon but still not able to scan any devices. App crashes in 2 seconds.
Which region do you apply. Would be very appreciated if you provide a simple code example that works (fufylev@gmail.com)

PlayCulture, 2023-01-25:
So with iBeacons you have to know the UUID you are looking for before you scan, you can't just scan for anything that's around. Here's a snipet from our project:

```
String Uuid = "00000000-0000-0000-0000-000000000000"; // Put your known UUID here
String Identifier = "com.playculture"; // Replace with your project ID
String Layout = "m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"; // iBeacon, must be this value
int ManufacturerId = 0x004c; // Reserved when using iBeacon, must be this value
```

For AltBeacon you can do all sorts of things, but iOS won't be able to transmit anything other than iBeacon.

Hope that helps!

dkozenkov, 2023-07-11:
> We just fixed our "problem" by using the solution from this [#566 (comment)](https://github.com/PhilipsHue/flutter_reactive_ble/issues/566#issuecomment-1181665284). 🎉 It is also outlined in the [...]

Thank you for your link. The problem is actually solved by making the specified changes to the file AndroidManifest.xml, and I used the string 'tools:targetApi="s"'. It is important to note that to start scanning, you need to request the necessary permissions directly in the application. In my case, I requested the following permissions: locationWhenInUse, bluetoothScan, bluetoothConnect. For this, I used the 'permission_handler' package from pub.dev.

closed 2023-08-29 by Taym95