PhilipsHue/flutter_reactive_ble #668: Incomplete response when readCharacteristic on iOS

v11, 2023-01-06:
First of all, thank you very much for this library. 

**Describe the bug**
I have an app where i read a characteristic from a BLE device (ESP32). The BLE device is scanning wi-fi networks and provides this data over the characteristic. Via my app i read this characteristic.

Everything is working fine. But yesterday I have experienced that sometimes (in a case where a lot of networks are disovered), not the complete data gets transmitted to my app. In the data, then some closing brackets are missinng and the decoding fails.

I have debugged this issue with the BT inspector app and there the full data is available and can be read.

In the flutter_reactive_ble package it is stated the follwing:
```
/// Be aware that a read request could be satisfied by a notification delivered
/// for the same characteristic via [characteristicValueStream] before the actual
/// read response arrives (due to the design of iOS BLE API).
```
I am wondering if this is the issue. But i have no idea how to workaround.

Data i recieve:
```
[
{"macaddress":"24:6F:28:D1:5D:DC"},
{"ssid":"FotokiteOfficeGuest"},
{"ssid":"FotokiteOffice"},
{"ssid":"FotokiteFactory"},
{"ssid":"FotokiteOffice"},
{"ssid":"FotokiteOfficeGuest"},
{"ssid":"FotokiteFactory"},
{"ssid":"fkG072B_2"},
{"ssid":"fkG092B_2"},
{"ssid":"DIRECT-B4-HP M479fdw Color LJ"},
{"ssid":"FotokiteOfficeGuest"},
{"ssid":"FotokiteFactory"},
{"ssid":"fkG132B_2"},
{"ssid":"FotokiteOffice"},
{"ssid":"Street-Files Agency 2.4 GHz"},
{"ssid":"Salt_2GHz_736A13"},
{"ssid":"DIRECT-4C-HP ENVY 5540 series"},
```

Data i expect to recieve:
```
[
{"macaddress":"24:6F:28:D1:5D:DC"},
{"ssid":"fkG072B_2"},
{"ssid":"FotokiteOffice"},
{"ssid":"FotokiteFactory"},
{"ssid":"FotokiteOfficeGuest"},
{"ssid":"fkG132B_2"},
{"ssid":"FotokiteFactory"},
{"ssid":"FotokiteOfficeGuest"},
{"ssid":"fkG092B_2"},
{"ssid":"FotokiteOffice"},
{"ssid":"DIRECT-B4-HP M479fdw Color LJ"},
{"ssid":"FotokiteOffice"},
{"ssid":"FotokiteFactory"},
{"ssid":"FotokiteOfficeGuest"},
{"ssid":"UPCD5B6DA7"},
{"ssid":"UPC Wi-Free"},
{"ssid":"qwp-69288"},
{"ssid":"hrc-23816"}
]
```

This is how i read the data:
```
Future<void> readFromBluetoothDevice(DiscoveredDevice device) async {

    final characteristic = QualifiedCharacteristic(serviceId: serviceUuid, characteristicId: characteristicUuid, deviceId: device.id);

    final response = await flutterReactiveBle.readCharacteristic(characteristic);

    ...
```

**Smartphone / tablet**
 - Device: [iPhone X]
 - OS: [iOS 13]

**Peripheral device**
- ESP32

**Additional context**
- Android is working fine.
- If there are fewer networks (fewer data), everything is working fine.

Liberations, 2023-01-09:
ios传入的UUID格式是两位 9CF1 android的是00009cf1-0000-1000-8000-00805f9b34fb 我这边特别处理下貌似可以了

v11, 2023-01-09:
Not sure what you mean. Can i just switch to this shorter ID?

Liberations, 2023-01-10:
Uuid.parse("9CF1")

v11, 2023-01-11:
Now i get the following exception:
`_Exception (Exception: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "A service C201 is not found in the peripheral E8CB3A98-9F28-F36A-2DA5-BD8A951130E6 (make sure it has been discovered)"))`

Could you (or someone else) be a bit more specific? I have read that in IOS you need to have this 16bit UUID. But is it for the service or for the characteristic or both?

Here is my full code:
```
// Summary
// This view is using a bluetooth (ble) connection to get network names (ssid) from the device.

// Check status with checkBluetoothStatus(). Status needs to be ready.
// Start scanning for bluetooth devices with startBluetoothScan()
// Connect to the device with connectBluetoothDevice()
// Multiple bluetooth errors will be handled:
// - No bluetooth
// - No location services (Android only)
// - bluetooth not authorized

// Read from ble device with readFromBluetoothDevice()
// Decode data and save ssid into a list ssidList
// Show list of networks in the view

import 'dart:async';
import 'dart:convert';
import 'dart:io';
import 'package:flutter/material.dart';
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart';
import 'package:mynotes/components/spacer.dart';
import 'package:mynotes/views/setup/views/setup_03_wifi_password_view.dart';
import 'package:mynotes/views/setup/services/setup_constants.dart';
import 'package:mynotes/views/setup/services/ssid_name.dart';
import 'package:mynotes/views/setup/services/setup_services.dart';
import 'package:app_settings/app_settings.dart';
import 'package:page_route_transition/page_route_transition.dart';

class SetupWifiView extends StatefulWidget {
  final SetupType setupType;

  const SetupWifiView({
    super.key,
    required this.setupType,
  });

  @override
  State<SetupWifiView> createState() => _SetupWifiViewState();
}

class _SetupWifiViewState extends State<SetupWifiView> with WidgetsBindingObserver {
  // BLE: A Universally Unique Identifier (UUID) is a globally unique 128-bit (16-byte) number that is used to identify profiles, services, and data types in a Generic Attribute (GATT) profile.
  final Uuid serviceUuid = Uuid.parse("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
  final Uuid characteristicUuid = Uuid.parse("beb5483e-36e1-4688-b7f5-ea07361b26a8");

  // BLE: GUI state management: Location Services
  bool isBleStatusUnauthorized = false;

  // BLE: GUI state management: Location Services
  bool isLocationServicesDisabled = false;

  // BLE: GUI state management: Not used yet
  bool isBluetoothPoweredOff = false;

  // BLE: GUI state management: BLE device is connected
  bool isBluetoothConnected = false;

  // BLE: GUI state management: BLE device has sent Data
  bool hasBluetoothData = false;

  // BLE: GUI state management: BLE device has sent Data
  bool isBottomSheet = false;

[... 492 lines cut ...]
                                child: ListView.builder(
                                  padding: const EdgeInsets.all(0.0),
                                  itemCount: ssidList?.length,
                                  itemBuilder: (BuildContext context, int index) {
                                    return widgetWifiNetwork(ssidList?[index]);
                                  },
                                ),
                              ),
                            ),
                          ),
                        ],
                      ),
                    ),
                  ),
                ],
              ),
            ),
          ),
        ],
      ),
    );
  }
}

```

Liberations, 2023-01-29:
after  case DeviceConnectionState.connected: try to discoverServices,like below
```
var resUUID = await flutterReactiveBle.discoverServices(id);
 resUUID.forEach((element) {
            debugPrint("resUUID" + element.serviceId.toString());
   });
```

v11, 2023-02-13:
I think, the problem was related to a buffer which was too small. The JSON response did not have a complete body including closing brackets, which then failed on decoding.

closed 2023-02-13 by v11