PhilipsHue/flutter_reactive_ble #675: Flutter Android App, scan fails with "location Permission Missing (code 3)" error

sidprice, 2023-01-15:
I am just beginning a new app using flutter_reactive_ble and have run into an issue right away when trying to scan for devices. The app is being developed for Android platform using the VSCode extension:

`Flutter 3.3.10 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 135454af32 (4 weeks ago) • 2022-12-15 07:36:55 -0800
Engine • revision 3316dd8728
Tools • Dart 2.18.6 • DevTools 2.15.0`

and flutter_reactive_ble v5.0.3.

This is in my Android Manifest:

``` 
        <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
        <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
        <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
        <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
```

The "scanForDevices" method returns an error; "Location Permission Missing (code 3)".

I also tried the alternate permission setting for BLUETOOTH_SCAN as suggested in the library readme.

```
       <uses-permission android:name="android.permission.BLUETOOTH_SCAN" 
                     tools:remove="android:usesPermissionFlags"
                     tools:targetApi="s" />
```

This gives the following build error:

```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:generateReleaseBuildConfig'.
> Error while evaluating property 'namespace' of task ':app:generateReleaseBuildConfig'
   > Failed to calculate the value of task ':app:generateReleaseBuildConfig' property 'namespace'.
      > Failed to query the value of property 'namespace'.
         > org.xml.sax.SAXParseException; systemId: file:/D:/DataRoot/Projects/Equla/ble_flutter/flutter_uart_sjp/android/app/src/main/AndroidManifest.xml; lineNumber: 5; columnNumber: 44; The prefix "tools" for attribute "tools:remove" associated with an element type "uses-permission" is not bound.
```

Suggestions would be most welcome.

Taym95, 2023-01-16:
Are you asking user to accept permission? scan will fail if you don't accept permission, try to use package like `permission_handler ` ? I don't think think this is library issue I will close it

closed 2023-01-16 by Taym95

sidprice, 2023-01-16:
@Taym95 You are so correct, thank you.