PhilipsHue/flutter_reactive_ble #683: how to re-generate device_interaction_tab.g.dart ?

mw66, 2023-02-02:
Hi,

I modified device_interaction_tab.dart (of the example dir), I'm wondering how to regenerate device_interaction_tab.g.dart?

I'm using AndroidStudio 4.2.1, strangle there is no "rebuild project" or "clean" under the "Build" menu.

Any way to do it via command line?

Thanks!

farr64, 2023-02-02:
@mw66: "Cleaning the build output files" is not obvious, but it's there under **Tools -> Flutter**:

[image]

This will **not** regenerate anything for you. It will simply **clean (delete)** the output of the build processes.

I don't know what you mean by "regenerate" something that you modified. If you need to go back to a **previous** version  (before you modified it), you probably need to retrieve it from your backup system -- or manually type, or use `undo` with the editor that did the changes for you.

Enjoy

mw66, 2023-02-02:
@farr64 

Thanks.

> I don't know what you mean by "regenerate"

https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/example/lib/src/ui/device_detail/device_interaction_tab.g.dart#L1

This file itself is generated (but I don't know where in the build process).

Taym95, 2023-02-02:
`flutter pub get && flutter pub run build_runner build --delete-conflicting-outputs`

closed 2023-02-02 by Taym95