PhilipsHue/flutter_reactive_ble #684: Subscribing to multiple characteristics in multiple services

vedangdate, 2023-02-02:
The scenarios that are currently working for me

1. subscribe to a Char C11 in Service S1.
2. subscribe to Char C11,C12,C13 in Service S1.

But when I subscribe to the characteristics of 2 different services, I am only seeing a consistent stream of a single characteristic in each of the services.
for e.g.
S1 has C11, C12, C13
S2 has C21, C22, C23

When I subscribe to all 6 chars I can only see C11 and C21 with consistent streaming data. All others either halt or don't show the required data.
When I subscribe to all 3 chars of a single service, all 3 have a consistent stream.

Any idea why this is happening

I am on version 5.0.3

my flutter version
Flutter (Channel stable, 3.3.10, on Microsoft Windows [Version 10.0.22621.1105], locale en-IN)

curiousmuch, 2023-02-12:
Do any of the UUIDs for C[11,12,13,21,22,23]  repeat?

Taym95, 2024-05-22:
This should work, I dont think this is library issues

closed 2024-05-22 by Taym95