PhilipsHue/flutter_reactive_ble #686: Characters returned on iOS are limited to 182 characters

abneradekz, 2023-02-07:
**Describe the bug**
On Android it is loading returning a List<int> with 210 to 240 depending on the moment.

On iOS this same return only returns 182 characters.

Follow the code below, can someone help me to know if I did something wrong or if there is an error?

**To Reproduce**
Steps to reproduce the behavior:

    Uuid serviceUuid = Uuid.parse("0000FFF0-0000-1000-8000-00805F9B34FB");
    Uuid characteristicUuid = Uuid.parse("0000FFF1-0000-1000-8000-00805F9B34FB");

    await flutterReactiveBle.deinitialize();

    await Future.delayed(const Duration(seconds: 1));

    await flutterReactiveBle.initialize();

    await Future.delayed(const Duration(seconds: 1));

    final deviceStreamActive = flutterReactiveBle.connectToDevice(id: global.device!.id);

    await Future.delayed(const Duration(seconds: 1));

    deviceStream = deviceStreamActive.listen((status) {
      print(status);
      if (DeviceConnectionState.connecting == status.connectionState) {
        print("connecting status");
      }

      if (DeviceConnectionState.connected == status.connectionState) {
        print("connected status");

        if (sub3 == null) {
          final characteristic = QualifiedCharacteristic(
            serviceId: serviceUuid,
            characteristicId: characteristicUuid,
            deviceId: deviceId,
          );

          await Future.delayed(const Duration(seconds: 1));

          sub3 = flutterReactiveBle.subscribeToCharacteristic(characteristic)
              .listen((data) async {
            print(data); // Android return 203 or more characters, but iOS return 182 characters - return List<int>
          }, onError: (dynamic error) {
            print("Subscribe error");
            print(error);
          });
        }
      } else {
        modelStatus.value = status;
      }
    });

**Smartphone / tablet**
 - Device: iPhone 6s
 - OS: iOS 15.5
 - Package version: 5.0.3

**Peripheral device**
- Vendor, model: iPhone 6s
- Does it run a custom firmware / software: no

iot49, 2023-02-09:
That’s an iPhone limitation

abneradekz, 2023-02-09:
I saw that iOS 15 supports up to 512 characters, but I already looked and I didn't find any place talking about it directly, can you guide me if there's a place that talks about it?

On the iPhone 6s can I change it to 512 if I have iOS 15?

How can I set the plugin to 512?

Here "https://github.com/nkolban/esp32-snippets/issues/325" talks about supporting it on iphone 7 and later, do you know of any place I can check and validate this?

abneradekz, 2023-02-14:
Good afternoon

I checked and the iPhone 6s has a limitation of 182 characters on receiving bluetooth data.

I tested it with the iPhone 11 and it worked, it returned 207 to 217 characters.

I couldn't test it on an iPhone 7 or 8, but I believe that everyone above these shouldn't have this limitation problem.

Thanks for the help and I'm closing the issue.

Hugs

closed 2023-02-14 by abneradekz