PhilipsHue/flutter_reactive_ble #705: MTU request inconsistency

pmagnuson, 2023-03-12:
**Describe the bug**
Trying to understand how the MTU request operates.  I have written a Flutter program which implements the Zephyr SMP protocol on a Nordic chip.  

The program performs the following MTU request and asks for 512 bytes:
```dart
 var mtu = await ble.requestMtu(deviceId: device.id, mtu: 512);
```
When I run this program against 2 different device FW versions, two different MTU results are reported which is understandable based on the configuration of the device FW.  

However, based on WireShark traces, the request from the Flutter program is different in the two cases.  
In one case the app request is 185 bytes:
[image]

In another case the app request is 527 bytes:
[image]

What factors on the flutter_reactive_ble library affect the actual Bluetooth MTU request?  
What can the Flutter app do to make this request larger?  

App Device:
iPhone 7
IOS 15.4.1
flutter_reactive_ble 5.0.3

pmagnuson, 2023-03-14:
More Information on this.  The Bluetooth device FW has been updated to provide a large MTU size of 498.  
For both Android and IOS, the following is used to request the MTU size
```dart
var mtu = await ble.requestMtu(deviceId: device.id, mtu: 512);
```
On Android, the MTU size is successfully negotiated and returns a value if 498.  
This shows the Wireshark trace where the App requests a MTU of 512, and the device responds with 498.

[image]

On IOS, the same function call is used to request the MTU with the same Bluetooth device and FW.  The returned value is 182.
The Wireshark trace shows that the IOS App is issuing a Bluetooth request for 182 bytes.  This happens when the requested MTU is 512, 400, 300.  

[image]

What can be done to raise the requested MTU size on IOS?  
App Device:
iPhone 7
IOS 15.4.1
flutter_reactive_ble 5.0.3

pmagnuson, 2023-03-27:
Thank you for looking at this.

closed 2023-03-27 by pmagnuson

j4m3s, 2023-03-27:
No worries. I also discovered something interesting this evening:  that if you use writeWithResponse() you can send the full number of MTU bytes to the server (from an android client anyway). But if you call writeWithoutResponse() the content is truncated to MTU-3 bytes. This latter appears to be what the spec says (though I didn't know it). What's odd is that writeWithResponse() actually sends the full bytes.

Ideally I think the library would throw an exception if you try to exceed the valid payload.