PhilipsHue/flutter_reactive_ble #708: Unneded string interpolation of characteristics is very resource hungry

bobatsar, 2023-03-20:
The logger output in `reactive_ble_mobile_platform.dart`, is evaluated for each received message. Especially the interpolation of the characteristic is quite resource hungry. We are receiving a lot of characteristics data and this string interpolation takes up about 10-15% of CPU usage for our data handling.

https://github.com/PhilipsHue/flutter_reactive_ble/blob/9aa9957e0d05c05362dc5553bc75c1f206417c1c/packages/reactive_ble_mobile/lib/src/reactive_ble_mobile_platform.dart#L59-L61

Unfortunately the logger is always initialized via the following code and I don't see any way to disable it.
https://github.com/PhilipsHue/flutter_reactive_ble/blob/9aa9957e0d05c05362dc5553bc75c1f206417c1c/packages/flutter_reactive_ble/lib/src/reactive_ble.dart#L100-L105

Can this be refactored, that the logger does not have to be initialized and can be set to `null` if not used? AFAIK then the interpolation would not be executed, resulting in less CPU usage.

Taym95, 2024-06-04:
if you set `logLevel` to `LogLevel.none` you will disable logs

bobatsar, 2024-06-05:
Hi @Taym95, 

the issue is, that even when setting the `logLevel` to `LogLevel.none` the string interpolation still happens.
There is nothing is logged in the end but the calculations are already done. When the logger is set to `null` this does not happen.

Can you please reopen this ticket and also think about merging my PR #861?

BenelliFurtado, 2024-06-05:
I noticed the same issue a few lines below. Our app is permanently scanning for Bluetooth devices and this is actually a big part of the used CPU!

https://github.com/PhilipsHue/flutter_reactive_ble/blob/9aa9957e0d05c05362dc5553bc75c1f206417c1c/packages/reactive_ble_mobile/lib/src/reactive_ble_mobile_platform.dart#L70-L72

bobatsar, 2024-06-05:
Hi @BenelliFurtado, yes that's the same issue. I am currently working on it and it seems my PR will be merged soon.

BenelliFurtado, 2025-06-10:
Hello, are there any news on the PR #861 ?