PhilipsHue/flutter_reactive_ble #722: Add a Kotlin version of handling BLE undeliverable exceptions

ElHyperion, 2023-04-20:
**Is your feature request related to a problem? (please describe)**
**How to handle the BLE undeliverable exception** sample code missing a Kotlin implementation.

**Describe the solution you'd like**
A Kotlin implementation for MainActivity.kt that works the same as the Java code in readme FAQ.

I am struggling with UndeliverableException errors that crash my app very frequently while reading characteristics (such as CHARACTERISTIC_READ), with no way to catch them in Flutter. I tried to rewrite the Java example code to Kotlin but it made my app unlaunchable.

I think that a verified Kotlin example should be added for people with no Kotlin knowledge like me, whose MainActivity is in Kotlin.

ElHyperion, 2023-04-25:
This is the code that I tried adding to MainActivity.kt:
```kotlin
import androidx.annotation.NonNull
import com.polidea.rxandroidble2.exceptions.BleException
import io.flutter.embedding.android.FlutterActivity
import io.flutter.embedding.engine.FlutterEngine
import io.reactivex.exceptions.UndeliverableException
import io.reactivex.plugins.RxJavaPlugins

class MainActivity : FlutterActivity() {
    override fun configureFlutterEngine(@NonNull flutterEngine: FlutterEngine) {
        RxJavaPlugins.setErrorHandler { throwable: Throwable ->
            if (throwable is UndeliverableException && throwable.cause is BleException) {
                return@setErrorHandler // ignore BleExceptions since we do not have subscriber
            } else {
                throw throwable
            }
        }
    }
}
```

It fails when I call ``Hive.initFluitter()``, with the following exception:
``Unhandled Exception: PlatformException(channel-error, Unable to establish connection on channel., null, null)``
The exception is thrown by ``_platform.getApplicationDocumentsPath();`` in path_provider.

remonh87, 2023-07-14:
the code is in kotlin.I literally copied it in my activity without issues.

```kotlin
 override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)

        RxJavaPlugins.setErrorHandler { throwable ->
            if (throwable is UndeliverableException && throwable.cause is BleException) {
                return@setErrorHandler // ignore BleExceptions since we do not have subscriber
            } else {
                throw throwable
            }
        }
    } 
```

closed 2023-07-14 by remonh87