PhilipsHue/flutter_reactive_ble #724: discoverServices throws error on iOS (only) -> PlatformException(reactive_ble_mobile.Central.(unknown context at $101e5b3c0).Failure:1

keehoo, 2023-04-23:
DiscoverServices functionality throws exception on iOS, works well on android. 

Steps to reproduce the behavior:
1. Scan for devices
2. call `    await flutterReactiveBle.discoverServices(device.id).then((value) {
      print(value);
    }).onError((error, stackTrace) {
      print(error);
    });`

3. Observe a failure :
4. [  +77 ms] flutter: PlatformException(reactive_ble_mobile.Central.(unknown context at $101e5b3c0).Failure:1, The operation couldn’t be completed. (reactive_ble_mobile.Central.(unknown context at
$101e5b3c0).Failure error 1.), {}, null)

Expected to recieve a list of `DiscoveredService`

Works on Android (Android 10)
```

// ANDROID
[  +29 ms] I/flutter ( 9929): [DiscoveredService(serviceId: 00001801-0000-1000-8000-00805f9b34fb, characteristicIds: [00002a05-0000-1000-8000-00805f9b34fb], characteristics:
[DiscoveredCharacteristic(characteristicId: 00002a05-0000-1000-8000-00805f9b34fb, serviceId: 00001801-0000-1000-8000-00805f9b34fb)], includedServices: []), DiscoveredService(serviceId:
00001800-0000-1000-8000-00805f9b34fb, characteristicIds: [00002a00-0000-1000-8000-00805f9b34fb, 00002a01-0000-1000-8000-00805f9b34fb, 00002a04-0000-1000-8000-00805f9b34fb], characteristics:
[DiscoveredCharacteristic(characteristicId: 00002a00-0000-1000-8000-00805f9b34fb, serviceId: 00001800-0000-1000-8000-00805f9b34fb), DiscoveredCharacteristic(characteristicId:
00002a01-0000-1000-8000-00805f9b34fb, serviceId: 00001800-0000-1000-8000-00805f9b34fb), DiscoveredCharacteristic(characteristicId: 00002a04-0000-1000-8000-00805f9b34fb, serviceId:
00001800-0000-1000-8000-00805f9b34fb)], includedServices: []), DiscoveredService(serviceId: 912ffff0-3d4b-11e3-a760-0002a5d5c51b, characteristicIds:
[  +61 ms] D/BluetoothGatt( 9929): onConnectionUpdated() - Device=00:22:36:11:BC:10 interval=36 latency=0 timeout=500 status=0
[+7664 ms] Application finished.

// IOS
[  +77 ms] flutter: PlatformException(reactive_ble_mobile.Central.(unknown context at $101e5b3c0).Failure:1, The operation couldn’t be completed. (reactive_ble_mobile.Central.(unknown context at
$101e5b3c0).Failure error 1.), {}, null)```

- device that works with other BLE libraries.

I'm trying to rewrite my app using this library.

DaemonChoejur, 2023-04-24:
This issue is sort of related to [https://github.com/PhilipsHue/flutter_reactive_ble/issues/385](385)
Please add a short delay before calling the discoverServices. 
Hope that helps.

keehoo, 2023-04-24:
@DaemonChoejur - thanks for the reply. Unfortunately that didn't help. 
[image]

At some point gave it 10 seconds.
```    print("Discovering services");
    await Future.delayed(const Duration(seconds: 10));
    await flutterReactiveBle.discoverServices(device.id).then((value) {
      print(value);
    }).onError((error, stackTrace) {
      print(error);
    });

ozzy1873, 2023-04-29:
This is suddenly happening to me too. Have you been able to solve it?

chipweinberger, 2023-05-01:
I'm hitting the same issue, but I have connected already:

```
  Future<void> connect() async {
    print("[BLE] ${_device.name}: connecting...");

    _connection = _ble.connectToDevice(
      id: _device.id,
      connectionTimeout: const Duration(seconds: 10),
    );

    print("[BLE] ${_device.name}: awaiting connection...");
    await _connection!.where((event) => event.connectionState == DeviceConnectionState.connected).first;
    print("[BLE] ${_device.name}: success! connected.");

    print("[BLE] ${_device.name}: discovering services...");

    // Discover services and characteristics
    final services = await _ble.discoverServices(_device.id.toString());
...
}
```

chipweinberger, 2023-05-01:
So something interesting.  I added a 3 second delay after connection, as suggestion.

It did not fix the problem, **_because the connection is immediately being disconnected after connection._** I'm not sure why, but my BLE peripheral makes a chime noise on disconnection so I know this is happening.