PhilipsHue/flutter_reactive_ble #725: [Android] Connecting after disconnecting problems

maciej-szulc, 2023-04-26:
**Describe error**.
[Android only, all versions - on iOS it works fine for me]

After disconnecting (canceling the stream), when I try to reconnect, the status remains "connecting" indefinitely. BUT there are two scenarios: 

1. this happens to me on the same device that was connected. That is, I can no longer connect to the same device, but I can connect to another device. 
2. it happens on another device. I am able to connect the same device, but I am not able to connect a new device. 

This only happens when I cancel the stream to disconnect the device. When my device is manually disconnected and the app has to reconnect, there are no problems. But when I disconnect any device by cancelling the stream, then errors occur.

Does anyone have any idea? 

**To reproduce**.
Steps to reproduce the behavior:
1. connect to the device. 
2. disconnect by canceling the stream. 
3. try again to connect to the same/another device. 

**Expected behavior**.
When I disconnect from any device, the app should be ready to reconnect to any new/same device. 

**Smartphone / tablet**.
 - Samsung Galaxy A52s
 - Android 13 (but also other versions, same situation).

**Peripheral device**.
- Custom device with custom software. 
**Additional**.

This is how I disconnect to device. 
[image]

maciej-szulc, 2023-04-27:
#301 I think it's the same issue.

maciej-szulc, 2023-04-27:
I think I have found a solution. Actually it was also proposed in #301 . I inserted a long delay after canceling the stream and before taking any other action. 
[image]

William-Geng, 2023-06-14:
I agree, for me instead of just wait a fixed amount of time, I check the connectedDeviceStream and wait for the disconnected event for that specific device, just do a firstwhere and check device ID match and state is "disconnected"

pawlowskim, 2024-02-01:
For me, the problem occurs only on Android. Stream returned from the connect method never fires `disconnect` event. I had to change the listener for  `connectedDeviceStream` as @William-Geng suggested. So I just ignored events from connect method and only used the subscription to disconnect from the device.

Moreover, I have faced issues with asynchronous actions, also only on Android. After disconnecting, I call some writes on the device, which creates unhandled connection that is never terminated (requires application to be killed, BT turned off / on on the device or restart the device manually). More here: https://github.com/PhilipsHue/flutter_reactive_ble/issues/589

rsubramanyam2994, 2024-07-24:
+1.  I have set a connection timeout of 1 second. Even after inserting a delay of 4 seconds between disconnecting device 1 and connecting to device 2, I am facing the same issue.  @maciej-szulc Is the delay working consistently for you?

maciej-szulc, 2024-07-24:
@rsubramanyam2994 I haven't tested it recently. Have you tried the more elegant solution that William suggested above?

closed 2025-08-09 by maciej-szulc