PhilipsHue/flutter_reactive_ble #728: [iOS] PlatformException "The operation couldn’t be completed." when calling connect

chipweinberger, 2023-05-01:
Exception has occurred.
PlatformException (PlatformException(reactive_ble_mobile.Central.(unknown context at $10330b3c0).Failure:1, The operation couldn’t be completed. (reactive_ble_mobile.Central.(unknown context at $10330b3c0).Failure error 1.), {}, null))

[image]

chipweinberger, 2023-05-01:
Looks similar to:

- https://github.com/PhilipsHue/flutter_reactive_ble/issues/724
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/632
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/624
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/516
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/223
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/573
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/339
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/251
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/137
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/205
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/95
- https://github.com/PhilipsHue/flutter_reactive_ble/issues/68

devtyagi-c, 2024-09-04:
I have the same issue
I perform these operations:
- Connect to Device, move to a new screen
- Click on start monitoring button
- On the click of the button the following operations are performed
    1. Enable notifications on characteristic 86f66001-f706-58a0-95b2-1fb9261e4dc7
    2. Enable notifications on characteristic 86f65002-f706-58a0-95b2-1fb9261e4dc7
    3. Write 0x0213C87462 on characteristic 86f65001-f706-58a0-95b2-1fb9261e4dc7
    4. Write 0x66 on characteristic 86f65001-f706-58a0-95b2-1fb9261e4dc7
   
Here is my code for the start monitoring button:
```dart
Future<void> startMonitoring() async {
    if (state.selectedDeviceId == null) {
      emit(state.copyWith(error: "No device selected"));
      return;
    }

    if (state.connectionState != DeviceConnectionState.connected) {
      emit(state.copyWith(error: "Device not connected"));
      return;
    }

    // Emit initial state
    emit(state.copyWith(isSettingUpMonitoring: true, error: null));

    try {
      await Future.delayed(const Duration(seconds: 2));

      // Enable notifications on 6001
      _monitorSubscription6001 = _monitorCharacteristic
          .execute(
        state.selectedDeviceId!,
        "86f66000-f706-58a0-95b2-1fb9261e4dc7",
        "86f66001-f706-58a0-95b2-1fb9261e4dc7",
      )
          .listen(
        (data) {
          // Emit state when a new value is received from 6001
          log("Received data from 6001: $data");
          emit(state.copyWith(characteristicValue: data.toString(), isMonitoring: true, isSettingUpMonitoring: false));
        },
        onError: (error) {
          emit(state.copyWith(error: error.toString(), isMonitoring: false, isSettingUpMonitoring: false));
        },
      );

      await Future.delayed(const Duration(milliseconds: 200));

      // Enable notifications on 5002
      _monitorSubscription5002 = _monitorCharacteristic
          .execute(
        state.selectedDeviceId!,
        "86f65000-f706-58a0-95b2-1fb9261e4dc7",
        "86f65002-f706-58a0-95b2-1fb9261e4dc7",
      )
          .listen(
        (data) {
          // We do not need to emit the state here when a new value is received from 5002
          log("Received data from 5002: $data");
        },
        onError: (error) {
          emit(state.copyWith(error: error.toString(), isMonitoring: false, isSettingUpMonitoring: false));
        },
      );

      await Future.delayed(const Duration(milliseconds: 200));

      await _writeCharacteristic.execute(
        state.selectedDeviceId!,
        "86f65000-f706-58a0-95b2-1fb9261e4dc7",
        "86f65001-f706-58a0-95b2-1fb9261e4dc7",
        [0x02, 0x13, 0xC8, 0x74, 0x62],
      );

      log("Successfully wrote data to 86f65001-f706-58a0-95b2-1fb9261e4dc7");

      await _writeCharacteristic.execute(
        state.selectedDeviceId!,
        "86f65000-f706-58a0-95b2-1fb9261e4dc7",
        "86f65001-f706-58a0-95b2-1fb9261e4dc7",
        [0x66],
      );

      log("Successfully wrote data to 86f65001-f706-58a0-95b2-1fb9261e4dc7");
    } catch (e) {
      // Emit state when an error occurs
      log("Error: $e");
      emit(state.copyWith(
        error: e.toString(),
        isMonitoring: false,
        isSettingUpMonitoring: false,
      ));
    }
  }
```

And I get this error: 
```
PlatformException (PlatformException(reactive_ble_mobile.Central.(unknown context at $10590a818).Failure:1, The operation couldn’t be completed. (reactive_ble_mobile.Central.(unknown context at $10590a818).Failure error 1.), {}, null))
```

[image]

[image]

Is there any fix for this?