PhilipsHue/flutter_reactive_ble #739: Discovering services after reconecting

AurimassKl, 2023-05-19:
When device is connected and when it lose connection it tries to connect to device again automatically and read services. but i get [Error occured when discovering services: PlatformException(service_discovery_failure, null, null, null)] error couple times but at the end it finds the way to work. Where could be the problem so it would't throw exeptions. 
Here is my code:

class BleConnect extends ConsumerWidget {
  @override
  Widget build(BuildContext context, WidgetRef ref) {
    final deviceConnectorP = ref.watch(connectorProvider);
    final connectionStateUpdateP = ref.watch(connectionStateProvider);
    final serviceDiscovererP = ref.watch(serviceDiscovererProvider);
    final device = ref.watch(deviceProvider);
    return connectionStateUpdateP.when(
        data: (data) {
          return _BleConnector(
              viewModel: DeviceInteractionViewModel(
                  deviceId: device.id,
                  connectionStatus: data.connectionState,
                  deviceConnector: deviceConnectorP,
                  discoverServices: () => serviceDiscovererP.discoverServices(device.id)));
        },
        error: ErrorScreen.new,
        loading: () => const LoadingScreen(page: 'device_detail'));
  }
}

class _BleConnector extends ConsumerStatefulWidget {
  const _BleConnector({
    required this.viewModel,
  });

  final DeviceInteractionViewModel viewModel;

  @override
  ConsumerState<_BleConnector> createState() => _BleConnectorState();
}

class _BleConnectorState extends ConsumerState<_BleConnector> {
  // final discoveredServices = StateProvider<List<DiscoveredService>>((ref) {
  //   return [];
  // });

  @override
  void initState() {
    if (!widget.viewModel.deviceConnected) {
      widget.viewModel.connect().then((value) => discoverServices());
      discoverServices().toString();
    }
    super.initState();
  }

  void RetryConnect() async {
    // widget.viewModel.disconnect();
    await widget.viewModel.connect().then((value) {
      // ref.watch(discoveredServices.notifier).state = [];
      discoverServices().toString();
    });
  }

  Future<void> discoverServices() async {
    await widget.viewModel.discoverServices();
    // ref.watch(discoveredServices.notifier).state = result;
  }

  @override
  Widget build(BuildContext context) {
    if (widget.viewModel.connectionStatus == DeviceConnectionState.connected) {
      return WifiScreen();
    } else {
      debugPrint("+RECONNECTING+");
      sleep(const Duration(seconds: 1));
      RetryConnect();
      return LoadingScreen(page: "Connecting to the device");
    }
  }
}

Mx-Tech, 2023-07-05:
I am facing the same issue. Is it a timing problem? Maybe the reconnection take some more time until it's finished and therefore the service discovery fails?
Can't test it currently.

remonh87, 2023-07-14:
Is this android or iOS ? If this is android it is recommended to make sure to keep a couple of seconds delay between connecting again. This is to give the ble stack enough time to settle down and get rid of flaky behaviour.

remonh87, 2023-08-07:
closed due to inactivity feel free to reopen or add a new issue if the problem persists

closed 2023-08-07 by remonh87