PhilipsHue/flutter_reactive_ble #740: Unable to scan if i provide UUID

anirudhsharma392, 2023-05-31:
I am unable to scan ibeacon if I provide UUID inside services, I think I am missing something. 
On the other hand, it's working fine if the empty array is provided to withServices.
here's my piece of code : 
```
    _subscription = await FlutterReactiveBle().scanForDevices(
        withServices: [Uuid.parse('11111111-1111-1111-1111-111111111111')],
        scanMode: ScanMode.lowLatency,
        requireLocationServicesEnabled:
            true).listen(onDeviceFound, onError: onError);
```
and here's the simulator screenshot :

[image]

paulobikiwelu, 2023-06-04:
@anirudhsharma392 same problem with me..
With Uuid .. nothing is found 😉

charismatest, 2023-07-07:
Same problem here, it's easier to scan without specifying services ans filter afterwards..

meuhle, 2023-07-11:
I'm working on an app that does the same thing, but it's working for me, are u sure u're specifying the right service UUID? and not the charateristic one?

You should also provide information about your devices, Phone/Tablet and peripheral

charismatest, 2023-07-12:
> I'm working on an app that does the same thing, but it's working for me, are u sure u're specifying the right service UUID? and not the charateristic one?
> 
> You should also provide information about your devices, Phone/Tablet and peripheral

The problem is, the device I'm working with doesn't advertise its services and I thought I could scan for the device UUID instead.

meuhle, 2023-07-12:
@charismatest try to print ad use the device.serviceUuids that it's printed, otherwise you can check for other paramteres of the device but inside the onData function handler, there you can check for the name of the device for example

```
flutterReactiveBle.scanForDevices(
            requireLocationServicesEnabled: false,
            withServices: [serviceUuid]).listen((device) {
                 device.name == 'YOURNAME' ? //do something : //do nothing
});
```

remonh87, 2023-07-14:
It looks like you are scanning using the device uuid on iOS. keep in mind that there is a difference between the UUID of the device (which is an iOS only generated id for the device) and the uuid of the service (which is one of the gatt services the device has).

closed 2023-07-14 by remonh87