PhilipsHue/flutter_reactive_ble #742: Extended advertisements not received by all Android devices (>= Bluetooth 5.0)

Hoiss, 2023-06-07:
**Describe the bug**
Extended advertising messages are not received by some Android devices.

**To Reproduce**
1.) Get a device that periodically sends extended advertisements.
2.) Use flutterReactiveBle.scanForDevices(withServices: [], scanMode: ScanMode.lowLatency, requireLocationServicesEnabled: true).listen((device) to receive advertisements on any Android12 (e.g.) device.
3.) There will be devices that do not show the extended advertisement messages (e.g. Redmi 10C, Xiaomi Mi A3). Other devices using the same app work flawlessly and show the extended advertising messages (e.g. OnePlus7T, Xiaomi Redmi A1, ...). The devices that do not show the extended advertising messages are able to receive legacy advertising messages.

**Expected behavior**
The same behavior on all devices with Bluetooth version >= 5.0 with the ability to receive extended advertisements.

**Smartphone / tablet**
 - Device: Tested OK: OnePlus7T, Xiaomi Redmi A1, some HTC. Tested NOT OK: Redmi 10C, Xiaomi Mi A3.
 - OS: Android 12
 - Package version: flutter_reactive_ble ^5.0.3

**Peripheral device**
- Vendor, model: Own development
- Does it run a custom firmware / software: yes

**Additional context**
I could imagine it's a phone or vendor specific smartphone problem, but all used ones offer Bluetooth 5.0 which has to support extended advertisements according to the spec. As soon as the sending peripheral device switches to legacy advertising mode all phones can see the messages.
If there is no solution available:
I need to understand, why there are Bluetooth 5.0 phones that are not able to see extended advertisements and if it's possible to determine if a phone is able to see them or not.

remonh87, 2023-07-14:
Ble discovery is really tricky and the fact that you do not see these advertisements can be caused by many issues: most commonly is the interference, device firmware and distance between the device and peripheral. Unfortunately there is no possibility to determine whether or not the device can see them.

One trick that you can do (and probably already did) is add the location permission to the android manifest and enable it. On some phones (for example nexus) there are reports that it improves discovery. I have to close this ticket because there is nothing we can do.

closed 2023-07-14 by remonh87

Hoiss, 2023-07-14:
OK, thank your for the information. Yes, I indeed already enabled location permission.
It's really a pitty that it can't be determined if a phone is able to receive all kind of BLE 5.0 messages or not. 
How to explain to customers that using the same app that works on phone A does not work on phone B while both support BLE5.0 ? I think in the end unfortunately the App is bad and not the underlying hardware.
I assume there might be a problem on some phones with the implementation of the BLE stack, because legacy messages can be seen, just "new fancy" BLE 5 ones like extended adv. messages can't be seen. But in this case these models are not really BLE5.0 compatible and the manufacturers should have to work on it. OK, this won't happen.