PhilipsHue/flutter_reactive_ble #751: Subscribing to a charateristic for second time after disconnecting and reconnecting

meuhle, 2023-06-28:
Hi I'm making an app to interact with a device i built, everything works fine for the first connection, so the app load, I scan for the device and then connect to it and subscribe to its charateristic, but then when i disconnect and reconnect to the same device I'm unable to subscribe and get the data from the charateristic, despite I reinitialize all the BLE components

andrepura, 2023-06-30:
i had the same issue with reconnecting to a device (on android & ios). disconnect is not really working, therefore the conenction is already established and if you try to "reconnect" there will be a second connection to the device, see #265 . After reusing the established connection, the problem (in my case) was solved.

meuhle, 2023-06-30:
@andrepura in my case the disconnection works, because the other machine tell me that no device are connected , and then i can reconnect,but then the notification doesn't receive any events, as written in the documentation the subscription to a notification charateristic should be closed on disconnection, and I get this state since i do a switch case on connection state and print a message on disconnection, but it also receive a Bluetooth GATT error onthe notification, I think that if the disconnection was handled in the right way this message should not appear, so which is the correct method to disconnect?. I don't want to reuse the same connection but create a new one as should the workflow be. If I try to just cancel the stream connection I don't know how but the notification keep coming, to stop this I must deinitialize the BLE. As you told in #256 , my behavoir is different since if i disconnect and reconnect I'm not receivng multiple stream but none

Mx-Tech, 2023-07-05:
@meuhle 
Are you sure the reconnection works properly? Currently I'm getting a service discovery failure after reconnection, only on Android devices. Also my device I want to connect to tells me that there was a connection established (a led turns on).

meuhle, 2023-07-11:
@Mx-Tech Yes my connection works prorperly since the LED on device turn on at first and second connection while turn off after the disconnection, the only problem is that if I try to read from a streming charateristic, on first connection it says enabled:true, and then on second connetion the same charateristic show enabed:false, but if I try using any BLE reader i can read the notification. So the error might seem to be the same like if the charateristic is not seen anymore on second connection

meuhle, 2023-07-11:
@Mx-Tech @andrepura I tried to read some other example doing the subscribtion and now is working good you must use 
`late StreamSubscription<ConnectionStateUpdate> connection;
var deviceConnectionController = StreamController<ConnectionStateUpdate>();

connection = flutterReactiveBle.connectToAdvertisingDevice(
        id: device.id,
        prescanDuration: scanTime(),
        withServices: [serviceUuid, g.cmdChar, g.debChar]).listen((event) {
      deviceConnectionController.add(event);
      switch (event.connectionState) {.... //HANDLE THE STATE HERE }
`
and then when you need to disconnect 
`g.connection.cancel();
        deviceConnectionController.add(
          ConnectionStateUpdate(
            deviceId: g.device.id,
            connectionState: DeviceConnectionState.disconnected,
            failure: null,
          ),
        );`

closed 2023-07-11 by meuhle