PhilipsHue/flutter_reactive_ble #756: PR #750 broke semver

christopherfujino, 2023-06-30:
FYI https://github.com/PhilipsHue/flutter_reactive_ble/pull/750 broke semantic versioning for `package:reactive_ble_platform_interface`. It added a new required parameter to a public constructor https://github.com/PhilipsHue/flutter_reactive_ble/pull/750/files#diff-117b0d313b30781c2ea45d6a46d971fa7d021152363245729fd0d0a9a51fae8cR53, which is a breaking change and should only have been published with a major version bump (that is, to `package:reactive_ble_platform_interface` version `6.0.0`. Since a pre v6 release was published with this breaking change, now users of `package:flutter_reactive_ble` version `5.0.3` or earlier will get the new version of the interface since flutter_reactive_ble depends on the interface with a caret dependency.

For example: https://github.com/flutter/flutter/issues/129841

RtgrV, 2023-06-30:
@christopherfujino thanks for mentioning this, we are currently working on it.

mbaz2, 2023-06-30:
having this issue as well, was on ^5.0.3 and now can no longer launch on Android and my pubspec.lock is upgrading to 5.1.0

RtgrV, 2023-06-30:
I was facing the same issue but didn't realize what was going on until I read ur message. Thanks for reporting it! 
It should be resolved with version 5.1.1

Taym95, 2023-06-30:
It should be resolved with version 5.1.1, I will close this issue.

closed 2023-06-30 by Taym95