PhilipsHue/flutter_reactive_ble #761: AGP fails due to com.google.protobuf version

ozzy1873, 2023-07-05:
I am trying to upgrade my project to gradle 8 using AGP Upgrade Assistant, but I am getting the following error:

`Could not get unknown property 'source' for generate-proto-generateDebugProto of type org.gradle.api.internal.file.DefaultSourceDirectorySet.`

This seems to be related to [this issue](https://github.com/android/codelab-android-datastore/issues/65).

Would it be possible to upgrade com.google.protobuf from 0.8.17 to 0.9.1?

ened, 2023-07-05:
@ozzy1873 could you try this branch - https://github.com/PhilipsHue/flutter_reactive_ble/pull/763 ?

ozzy1873, 2023-07-06:
I am getting the same error. The pubspec reference was changed to:

  flutter_reactive_ble:
    git:
      url: https://github.com/ened/flutter_reactive_ble.git
      path: packages/flutter_reactive_ble

Is that correct?

ened, 2023-07-06:
Correct, but you'll need to change the `ref` as well. Use `b40a91dc6b5b8205e60eb6742ce73746b2283ef7`

ozzy1873, 2023-07-06:
Same error. Your gradle-wrapper.properties file still references 7.0.2. Is that correct?

ened, 2023-07-06:
I assume you refer to `packages/reactive_ble_mobile/android/gradle/wrapper/gradle-wrapper.properties` -> and that file should be removed (the entire `packages/reactive_ble_mobile/android/gradle` folder actually).

I think you need to refer to `reactive_ble_mobile`, as that is the android/iOS part, so:

```
reactive_ble_mobile:
  git:
    url: https://github.com/ened/flutter_reactive_ble.git
    path: packages/reactive_ble_mobile
    ref: 'b40a91dc6b5b8205e60eb6742ce73746b2283ef7'
```

ozzy1873, 2023-07-06:
That worked, thanks!

AndreaBalde, 2023-12-06:
I also try with the last commit: ```f52bd60e4425b460265bb04003b753d65a10af48```

But I've get this error:
```
Resolving dependencies...
Because flutter_reactive_ble 5.2.0 depends on reactive_ble_mobile ^5.2.0 and no versions of flutter_reactive_ble match >5.2.0 <6.0.0, flutter_reactive_ble ^5.2.0 requires reactive_ble_mobile from hosted.
So, because kubeuf depends on both flutter_reactive_ble ^5.2.0 and reactive_ble_mobile from git, version solving failed.

You can try the following suggestion to make the pubspec resolve:
* Consider downgrading your constraint on flutter_reactive_ble: flutter pub add flutter_reactive_ble:'^3.1.1+1'
exit code 1
```

riccardocescon, 2023-12-06:
I am getting the same issue.
@ozzy1873 does it still work for you? have you changes anything?
@ened I can read that the line on the error says `
Because flutter_reactive_ble 5.2.0 depends on reactive_ble_mobile ^5.2.0 and no versions of flutter_reactive_ble match >5.2.0 <6.0.0`
But the version is 5.2.0, shouldn't it be 5.2.1 with your fix on #763?

closed 2024-01-11 by Taym95

Hoiss, 2024-01-15:
I'm also getting 
**Could not get unknown property 'source' for generate-proto-generateDebugProto of type org.gradle.api.internal.file.DefaultSourceDirectorySet**. 

So I changed the pubspec to
` flutter_reactive_ble:
    git:
      url: https://github.com/PhilipsHue/flutter_reactive_ble.git
      ref: '129844e6e266bebe2ba0fe18f305db137e21c8a0'`
A `flutter pub get` now shows the error `Resolving dependencies... 
Error on line 1, column 7: "name" field doesn't match expected name "flutter_reactive_ble".
  ╷
1 │ name: flutter_reactive_ble_workspace
  │       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^`

But I can't find any flutter_reactive_ble_workspace in the files...
What am I doing wrong?

riccardocescon, 2024-01-15:
> I'm also getting **Could not get unknown property 'source' for generate-proto-generateDebugProto of type org.gradle.api.internal.file.DefaultSourceDirectorySet**.
>
> So I changed the pubspec to ` [...]

They are working on an update for the package to fix this issue, it should be updated next days i think

Hoiss, 2024-01-15:
Thank you @riccardocescon.
Isn't there a temporary workaround available in case this is a blocking issue?

riccardocescon, 2024-01-15:
> Thank you @riccardocescon. Isn't there a temporary workaround available in case this is a blocking issue?

You're welcome.
The solution i am going with is by downgrading the agp and work with Android 33, and update everything when they release this update