PhilipsHue/flutter_reactive_ble #768: App crash due to io.reactivex.exceptions.UndeliverableException while closing characteristic streams

ElHyperion, 2023-07-10:
**Describe the bug**
I'm frequently getting the following error when I temporarily lose connection to a device with a subscription to 3 characteristics and attempt to reconnect to it, immediately after I receive DeviceConnectionState.disconnecting:

```dart
D/BluetoothGatt(20541): setCharacteristicNotification() - uuid: 00003a02-0f1a-af1a-a1c7-9ec071e58224 enable: false
D/BluetoothGatt(20541): close()
D/BluetoothGatt(20541): setCharacteristicNotification() - uuid: 00003b01-0f1a-af1a-a1c7-9ec071e58224 enable: false
W/System.err(20541): io.reactivex.exceptions.UndeliverableException: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 0 (GATT_SUCCESS)
D/BluetoothGatt(20541): unregisterApp() - mClientIf=7
W/System.err(20541): 	at io.reactivex.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:367)
W/System.err(20541): 	at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.dispose(ObservableFlatMap.java:313)
W/System.err(20541): 	at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
... (skipping the rest of the stack trace)
E/AndroidRuntime(20541): FATAL EXCEPTION: main
E/AndroidRuntime(20541): Process: com.jetimodel.jetistudiomobile, PID: 20541
E/AndroidRuntime(20541): io.reactivex.exceptions.UndeliverableException: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 0 (GATT_SUCCESS)
E/AndroidRuntime(20541): 	at io.reactivex.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:367)
E/AndroidRuntime(20541): 	at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.dispose(ObservableFlatMap.java:313)
E/AndroidRuntime(20541): 	at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
... (skipping some part of the stack trace)
E/AndroidRuntime(20541): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
E/AndroidRuntime(20541): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
E/AndroidRuntime(20541): Caused by: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 0 (GATT_SUCCESS)
E/AndroidRuntime(20541): 	at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
E/AndroidRuntime(20541): 	at android.bluetooth.BluetoothGatt$1$5.run(BluetoothGatt.java:363)
E/AndroidRuntime(20541): 	at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:1003)
E/AndroidRuntime(20541): 	at android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback(Unknown Source:0)
E/AndroidRuntime(20541): 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:358)
E/AndroidRuntime(20541): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:135)
E/AndroidRuntime(20541): 	at android.os.Binder.execTransactInternal(Binder.java:1321)
E/AndroidRuntime(20541): 	at android.os.Binder.execTransact(Binder.java:1280)
Lost connection to device.
```

This error crashes my app too. Is there a way to handle this error, since it's clearly coming out of ?

**To Reproduce**
1. Connect to a device using `connectToDevice()`
2. Subscribe to protected characteristics
3. Power off the device
4. Try to connect back immediately

**Expected behavior**
The error either does not happen or is caught and logged nicely.

**Smartphone / tablet**
 - Device: Samsung Galaxy S20 FE 5G
 - OS: Android 13
 - Package version: [e.g. 5.1.1]

**Peripheral device**
- Vendor, model: ESP32
- Does it run a custom firmware: yes

remonh87, 2023-07-14:
This is a problem in rx java please check the [readme](https://github.com/PhilipsHue/flutter_reactive_ble#how-to-handle-the-ble-undeliverable-exception) how to fix it.

closed 2023-07-14 by remonh87

ElHyperion, 2023-07-14:
> 

The example in readme is for Java. My main activity is in Kotlin, and I haven't been able to successfully translate the Java code to Kotlin. I mentioned in #722

remonh87, 2023-07-14:
the example code is in kotlin you can just copy paste it