PhilipsHue/flutter_reactive_ble #770: Phone does not disconnect from peer if reconnecting after the disconnected event arrives

maBarabas, 2023-07-11:
**Describe the bug**

The phone did not disconnect from the remove device when calling disconnect.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device using `connectToAdvertisingDevice()`
2. Wait for the connected event
3. Disconnect
4. Wait for the disconnected event
5. Reconnect to the same device using `connectToAdvertisingDevice()`

Notice that the second connects completes very quickly. Also notice that as far as the peer device is concerned, we never disconnected from it. It seems like that when I received the disconnected event, the disconnect was still in progress, it was cancelled and the previous connection was reused.

**Expected behavior**

When I receive the disconnected event, then the phone has been disconnected from the peer.

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Pixel 5, Galaxy Note 10+, Xperia 5
 - OS: Android 11, 12, 13
 - Package version: 5.0.3

**Peripheral device**
- Vendor, model: custom
- Does it run a custom firmware / software: yes

**Additional context**

iPhone works as expected.

Gufdoor, 2023-08-02:
@maBarabas having the same issue here, but instead I'm using **connectToDevice()**

meuhle, 2023-08-03:
@Gufdoor I found theright way to handle the disconnection 
this is how i connect to the device
```
connection = g.flutterReactiveBle.connectToAdvertisingDevice(
        id: g.device.id,
        prescanDuration: scanTime(),
        withServices: [g.serviceUuid, g.cmdChar, g.debChar]).listen((event) {
      deviceConnectionController.add(event);
      switch (event.connectionState) {
```
And this is how i handle the disconnection
```
connection.cancel();
        deviceConnectionController.add(
          ConnectionStateUpdate(
            deviceId: g.device.id,
            connectionState: DeviceConnectionState.disconnected,
            failure: null,
          ),
```