PhilipsHue/flutter_reactive_ble #774: Request MTU and then `subscribeToCharacteristic` listener can't receive the notifications

wcoder, 2023-07-17:
**Describe the bug**

The `subscribeToCharacteristic` listener can't receive the notifications after MTU was requested.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device using `connectTo()` / `connectToAdvertisingDevice()`
2. Request MTU (512 - the device supports 498) -> `requestMtu` returns 498
3. Subscribe to a non-protected characteristic

**Actual behavior**
```
I/flutter ( 1897): REACTIVE_BLE: Start subscribing to notifications for QualifiedCharacteristic(characteristicId: XXX, serviceId: XXX, deviceId: XXX)
D/BluetoothGatt( 1897): setCharacteristicNotification() - uuid: XXX enable: true
```
and nothing.

**Expected behavior**
The `subscribeToCharacteristic` listener should receive the notification immediately after subscription.

- [x] I tried doing the same with nRF Connect (on Android and iOS) and it exhibits the expected behavior as described above
- [x] I tried doing the same with different Android util apps and it exhibits the expected behavior as described above

**Workaround**

The solution that can help resolve this bug: #773 

**Smartphone / tablet**
 - Device: Samsung, Xiaomi
 - OS: Android 10+
 - Package version: 5.0.3

**Peripheral device**
- Vendor, model: Custom, Zefyr BLE
- Does it run a custom firmware: yes

chplalex, 2023-08-03:
@wcoder 
I faced a similar issue but with another sign: the subscribeToCharacteristic listener received nothing without Request MTU (when the default MTU size is used). But, when I set the MTU to 247 (I requested 250), the listener began to receive values. 

The issue occurs on Android. On iOS all works great for me.

I think the specific subscribeToCharacteristic request requires the relevant MTU size. When the actual MTU size is not relevant to the specific request, the listener receives nothing.

Try to play with MTU size.

Gufdoor, 2024-01-03:
I faced the same issue and fortunately I found this topic. Tried requesting MTU and worked!

Taym95, 2024-04-08:
Make sure to request MTU then use `subscribeToCharacteristic `

closed 2024-04-08 by Taym95