PhilipsHue/flutter_reactive_ble #775: Exception when trying to write Characteristic with Response

RubenKid, 2023-07-22:
**Describe the bug**
Exception is raised when try to write credentials with Response using an iOS device

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one / many devices using `connectTo()` / `connectToAdvertisingDevice()`
2. write to non-protected characteristic
3. Observe a failure with below exception

Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "Error Domain=CBATTErrorDomain Code=14 "Unlikely error." UserInfo={NSLocalizedDescription=Unlikely error.}"

**Expected behavior**
Characteristic should be written

- [ ] I tried doing the same in Android and it works perfect. Also tried with Light Blue app and it woks

**Smartphone / tablet**
 - Device: iPhone 13
 - OS: iOS 16
 - Package version: 5.1.1

wcoder, 2023-07-22:
`CBATTErrorDomain Code=14 "Unlikely error."` looks like it's an iOS BLE error

- check your characteristic via iOS nRF Connect
- check your data length and used or not MTU

RubenKid, 2023-07-24:
Checked with iOS nRF Connect and It seems the device disconnects after writing, this could be a reason I believe

[image]