PhilipsHue/flutter_reactive_ble #777: Cannot Connect to more than 9 - Android 13

scott-olero, 2023-07-28:
**Describe the bug**
We have been able to connect to 12 slaves but after Android 13 update we can only get a maximum of 9 slaves.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to more than 9 slaves

**Expected behavior**
The 10th slave wont connect , seems to be blocked by the Android 13 OS.

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: [Samsung Active TAB 3]
 - OS: [ Android 13]
 - Package version: [e.g. 1.1]

**Peripheral device**
- Vendor, model: [ CUSTOM - Silicon labs bgm210]
- Does it run a custom firmware / software: yes

remonh87, 2023-08-07:
did you receive an error message in the logs? As far as I know we do not limit the amount of subsequent connections power I know that connecting to > 9 devices is a discouraged use case.

scott-olero, 2023-08-07:
Hi @remonh87 we did not get any errors . We optimized the app to connect reliably with 12 sensors, but its seems like the issue is an Android 13 OS issue rather than a Library issue.