PhilipsHue/flutter_reactive_ble #779: Connection problem Samsung S20 FE

Berkant35, 2023-08-06:
Hello, we have a mobile application, and this mobile app is designed to control multiple electronic devices over BLE. I'm experiencing some issues with this situation. While it works flawlessly with 6 devices that I want to connect to, it doesn't work properly on the Samsung S20 I've indicated below. As seen in the video, our customer can connect to 4 devices initially but fails to connect to 2 of them. However, when the application is closed and everything is refreshed, it can connect to all 6 devices seamlessly. Do you have any suggestions for a solution to this?

https://github.com/PhilipsHue/flutter_reactive_ble/assets/55640416/8d39cb7c-baf8-4811-84c9-6cb5ab1ce404

import 'dart:async';

import 'package:flutter/material.dart';
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart';
import 'package:flutter_riverpod/flutter_riverpod.dart';

import '../catchpad_flutter_lib_init.dart';
import '../provs/ble_prov.dart';
import 'device_model.dart';
import 'reactive_state.dart';

class BleScanner implements ReactiveState<BleScannerState> {
  BleScanner(this.ref) : _ble = ref.watch(bleProv);
  final Ref ref;

  final FlutterReactiveBle _ble;

  StreamController<BleScannerState>? _stateStreamController;
  bool _pushedStateOnce = true;

  StreamSubscription? _subscription;

  var _devices = <DeviceModel>{};
  final _lastdevices = <DeviceModel>{};

  @override
  Stream<BleScannerState> get state {
    return _stateStreamController!.stream;
  }

  void updateDeviceInfo(DeviceModel newDev) {
    final d = <DeviceModel>{};

    for (var dev in _devices) {
      if (dev.id == newDev.id) {

        d.add(newDev);
      } else {
        d.add(dev);
      }
    }
    _devices.clear();
    _devices = Set<DeviceModel>.from(d);

    _pushState();
  }

  void deleteDevices(List<String> deadList) async {
    pauseScan();
    _lastdevices.clear();
    for (var deadId in deadList) {
      if (_devices.map((e) => e.id).contains(deadId)) {
        _devices.removeWhere((element) => element.id == deadId);
      }
    }
    _pushState();
    resumeScan();
  }

  void refreshScan({required List<DiscoveredDevice> connectedDevices}) async {
    final lastdevices = Set<DiscoveredDevice>.from(_lastdevices);
    for (var condev in connectedDevices) {
      lastdevices.removeWhere((lastdev) => lastdev.id == condev.id);
    }
    final orgdevs = Set<DiscoveredDevice>.from(_devices);
    final intersections = lastdevices.intersection(orgdevs);
[... 392 lines cut ...]
          weight: 900
          style: italic
flutter_intl:
  enabled: true

flutter doctor
Please commit your changes or stash them before you merge.
Aborting
berkantcalikusu@Berkants-MacBook-Pro catchpad % flutter --version    
Flutter 3.10.5 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 796c8ef792 (8 weeks ago) • 2023-06-13 15:51:02 -0700
Engine • revision 45f6e00911
Tools • Dart 3.0.5 • DevTools 2.23.1
berkantcalikusu@Berkants-MacBook-Pro catchpad % flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.10.5, on macOS 13.4.1 22F770820d darwin-arm64, locale en-TR)
[✓] Android toolchain - develop for Android devices (Android SDK version 33.0.0-rc2)
[✓] Xcode - develop for iOS and macOS (Xcode 14.3.1)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2021.2)
[✓] Android Studio (version 2022.2)
[✓] IntelliJ IDEA Community Edition (version 2022.2.4)
[✓] VS Code (version 1.80.1)
[✓] Connected device (3 available)
[✓] Network resources

remonh87, 2023-08-07:
did you try using a bluetooth explorer app like Nrfconnect and try to connect to the devices. Also did you receive any errors in logcat?

I would be very cautious with connecting a lot of ble devices at the same time because the OS can always decide to disconnect because it needs the bandwidth for other connections (WIFI or BLE).

Berkant35, 2023-08-10:
> 

I didn't. Because this problem occurred on the customer's end, and this phone is located remotely from me. I am already working with my real devices smoothly.

"I would be very cautious with connecting a lot of ble devices at the same time because the OS can always decide to disconnect because it needs the bandwidth for other connections (WIFI or BLE)." 

Thank you for this advice. How can I manage to fix it? Should I add a delay or try something else?