PhilipsHue/flutter_reactive_ble #780: Thoughts on Bluetooth pairing

paintingStyle, 2023-08-07:
**Is your feature request related to a problem? (please describe)**
Hi，According to what I have investigated so far, iOS cannot get the device pairing status when the bluetooth device is not connected,
And Android can get the device pairing status, is this correct?

The Android code below:
```
import android.bluetooth.BluetoothAdapter;
import android.bluetooth.BluetoothDevice;

BluetoothAdapter bluetoothAdapter = BluetoothAdapter.getDefaultAdapter();
Set<BluetoothDevice> pairedDevices = bluetoothAdapter.getBondedDevices();

if (pairedDevices.size() > 0) {
    for (BluetoothDevice device : pairedDevices) {
        String deviceName = device.getName();
        String deviceHardwareAddress = device.getAddress(); 
        // pairing status
        int bondState = device.getBondState();
        
        switch (bondState) {
            case BluetoothDevice.BOND_BONDED: 
                Log.d("BondState", "BluetoothDevice.BOND_BONDED");
                break;
            case BluetoothDevice.BOND_BONDING:
                Log.d("BondState", "BluetoothDevice.BOND_BONDING");
                break;
            case BluetoothDevice.BOND_NONE: 
                Log.d("BondState", "BluetoothDevice.BOND_NONE");
                break;
        }
    }
}
```

**Describe the solution you'd like**
I'm wondering if I can get the pairing status of bluetooth devices on an android phone via [flutter_reactive_ble]?
If not I'll try to get it myself using the code above

Thanks

AymenXD, 2023-08-07:
Hey, I am encountering the same issue even with flutter_blue_plus. Please, if you found any work-arounds or any news concerning this matter, I would be grateful if you share them with us.

remonh87, 2023-08-07:
Thanks for reporting. The difficulty with Bluetooth is that both Android and iOs are a bit different. To make a good cross platform plugin we try to refrain from specific android or iOS functions because else it will make the usage of the plugin more complex. 

That is the reason why it is not implemented. I would indeed create a simple plugin yourself  that propagates the bonded state from the device.

closed 2023-08-07 by remonh87