PhilipsHue/flutter_reactive_ble #784: Android seems to initiate disconnect and GATT reports status=22 but no events in connection stream.

western-hoolock, 2023-08-21:
**Is your feature request related to a problem? (please describe)**
I have an Android 12 phone (Nokia 5.4) that connects to a custom peripheral and sometimes the connection is lost and the only information the Flutter logs show are:
```
D/BluetoothGatt(32648): onClientConnectionState() - status=22 clientIf=8 device=4C:12:AE:6C:A6:8F
D/BluetoothGatt(32648): close()
D/BluetoothGatt(32648): unregisterApp() - mClientIf=8
```

The peripheral side shows the connection has been dropped but I have not been notified of any event at all in the connection stream, my app thinks it's still connected to the peripheral. A bit of Googling reveals that GATT status 22 means the connection has been close from the Android side? Not sure why this would happen but numerous experiments revealed no pattern, sometimes it happens within minutes other times after 30 minutes.

Root of the problem could very well be in the peripheral side but I would still like to be notified of ```status=22``` through the connection stream because clearly the connection is invalid after that. (Characteristic reads start throwing ```GenericFailure``` exceptions after that ```status=22```)

**Describe the solution you'd like**
Ideally an event should be written to the connection stream to indicate the connection has been dropped/disconnected. 

**Describe alternatives you've considered**
Reconnecting after a generic exception when reading/writing is an option but not a very nice one.

**Additional context**

mqhamdam, 2023-09-19:
Same issue. In my case, app BLE device not discoverable if I don't disconnect from it manually. Have to turn off/on device, to connect to it again.

YasserDRIF, 2023-09-19:
Same issue here, Getting this error in the pairing process : 
 Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 22 (GATT_CONN_TERMINATE_LOCAL_HOST)

truedat101, 2025-12-18:
Not a huge deal ... but I'm curious if anyone has done a root cause analysis?  Seems to affect a specific device I have, didn't notice this in testing other android devices.