PhilipsHue/flutter_reactive_ble #785: Please support protobuf ^3.0.0

RubenGarcia, 2023-08-23:
**Is your feature request related to a problem? (please describe)**
flutter_reactive_ble does not support protobuf ^3.0.0

**Describe the solution you'd like**
protobuf ^3.0.0 is supported

**Describe alternatives you've considered**
Please consider indicating in
https://pub.dev/packages/flutter_reactive_ble/install
that protobuf 3 is not supported.

**Additional context**
```
$ dart pub add flutter_reactive_ble
The current Dart SDK version is 3.0.6.

Because flutter_reactive_ble >=3.0.0-nullsafety.0 <3.0.0 depends on protobuf ^2.0.0-nullsafety.0 and flutter_reactive_ble <3.0.0-nullsafety.0 doesn't support null safety, flutter_reactive_ble <=3.0.0-nullsafety.0 requires protobuf ^2.0.0-nullsafety.0.
And because flutter_reactive_ble ^3.0.0 depends on protobuf ^2.0.0 and flutter_reactive_ble >=4.0.0 <4.0.1 depends on reactive_ble_mobile ^4.0.0, flutter_reactive_ble <4.0.1 requires protobuf ^2.0.0-nullsafety.0 or reactive_ble_mobile ^4.0.0.
And because flutter_reactive_ble ^4.0.1 depends on reactive_ble_mobile ^4.0.1 and flutter_reactive_ble >=5.0.0 <5.0.1 depends on reactive_ble_mobile ^5.0.0, flutter_reactive_ble <5.0.1 requires protobuf ^2.0.0-nullsafety.0 or reactive_ble_mobile ^4.0.0 or ^5.0.0.
And because flutter_reactive_ble >=5.0.1 <5.0.2 depends on reactive_ble_mobile ^5.0.1 and flutter_reactive_ble >=5.0.2 <5.0.3 depends on reactive_ble_mobile ^5.0.2, flutter_reactive_ble <5.0.3 requires protobuf ^2.0.0-nullsafety.0 or reactive_ble_mobile ^4.0.0 or >=5.0.0 <6.0.0.
And because flutter_reactive_ble >=5.0.3 <5.1.0 depends on reactive_ble_mobile ^5.0.3 and flutter_reactive_ble >=5.1.0 <5.1.1 depends on reactive_ble_mobile ^5.1.0, flutter_reactive_ble <5.1.1 requires reactive_ble_mobile ^4.0.0 or >=5.0.0 <6.0.0 or protobuf ^2.0.0-nullsafety.0.
And because flutter_reactive_ble >=5.1.1 depends on reactive_ble_mobile ^5.1.1 which depends on protobuf ^2.0.0, every version of flutter_reactive_ble requires protobuf ^2.0.0-nullsafety.0.
So, because sdip_test_suite depends on both protobuf ^3.0.0 and flutter_reactive_ble any, version solving failed.

The lower bound of "sdk: '>=2.3.0-dev <3.0.0'" must be 2.12.0 or higher to enable null safety.
For details, see https://dart.dev/null-safety
```

Kiranpatilmb, 2023-09-06:
@RubenGarcia can use the downgraded version of protobuf to protobuf: ^2.1.0

RubenGarcia, 2023-09-07:
protobuf ^2.1.0 gives me
```
Error: Couldn't find constructor 'GrpcServiceName'.
```

Kiranpatilmb, 2023-09-07:
Please clean your project and after run flutter pub get it will work and if still it is not working try to clear cache of your project it will work. If it still not work ..try with 2.0.0

RubenGarcia, 2023-09-07:
GrpcServiceName is a grpc 3 class, it does not exist on grpc 2.0. or 2.1.
This is not urgent for us, we have a workaround.
Just consider supporting 3.0 in the future when you have time.

Kiranpatilmb, 2023-09-07:
Ohh, 
Okay. Will let you know!