PhilipsHue/flutter_reactive_ble #792: Question : How can I get the response from device?

expertdev-designer, 2023-09-01:
Hii all  
Thanks in advance if anyone helps me understand this I am new to developing the flutter application with Bluetooth as I don't have much knowledge about Bluetooth protocol. With the help of this package, I am able to implement scanning devices integration but could not figure out how I can get the response from the device when I have written the command.

**Additional context**
how I get a response like below

Device connection password：0x38 38 38 38
APP:0x8f 38 38 38 38 
Device response：0x8f Code 5F 56 32 2E 30

APP:0x8f 38 38 38 11 
Device response：0x8f 45 52 52 4F 52

Taym95, 2023-09-04:
I would advice you to read about Bluetooth protocol before using the library, there is something called listen to notification in Ble.

closed 2023-09-04 by Taym95

expertdev-designer, 2023-09-04:
@Taym95 Could you please share the link where I can easily understand