PhilipsHue/flutter_reactive_ble #793: Warning! No event channel set up to report a connection update

yxyyuanxy, 2023-09-06:
hello, I got Waring like "Warning! No event channel set up to report a connection update" when I connect to a device, but sometimes it won't appear again， What is the reason for this situation? How can I solve it?

IceTears1, 2023-09-09:
i,m same issues

nissaba, 2023-09-28:
getting the same, and I don't know if it as any thing to do with my problem. the stream for the connectToDevice does not update at all. the device does get connected as it will stop advertising. 

If I hot-reload app then I do get 1 connection update and that is it. I never get to have an update that I got connected. 

if I use the connectToAdvertisingDevice

I get the an infinite loop informing me that the device is not connected. 

flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c
flutter: ConnectionState for device 3d19f04a-1e9e-1406-68de-2b6085bf185c : DeviceConnectionState.disconnected
flutter: Device not connected
flutter: Start connecting to 3d19f04a-1e9e-1406-68de-2b6085bf185c

this as been a blocker for me for about a week now.

irfanismaya20, 2023-10-06:
Any update for "No event channel set up to report a connection update"? not working in iOS 15, iOS 17 but in iOS 12 running well